shivasiddharth/GassistPi #693: Radio and playing playlists from youtube does not work

SkyCZ, 2018-12-20:
### Board and OS details:    
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 2a02082
Serial		: 000000006b52d57a

```

```
 PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
Radio will not play and it will freeze the assistant, youtube playlist will not play only the 1st song of it
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation?
No

### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If the answer to question 4 is No, then proceed further.

### Paste the command below, that crashed the assistant
```
"Hey Google, play radio 1"
```

### Paste the contents of the terminal below
```
pro 20 18:40:47 raspberrypi python[19188]:     lang=detect(word)
pro 20 18:40:47 raspberrypi python[19188]:   File "/home/pi/env/lib/python3.5/si
pro 20 18:40:47 raspberrypi python[19188]:     return detector.detect()
pro 20 18:40:47 raspberrypi python[19188]:   File "/home/pi/env/lib/python3.5/si
pro 20 18:40:47 raspberrypi python[19188]:     probabilities = self.get_probabil
pro 20 18:40:47 raspberrypi python[19188]:   File "/home/pi/env/lib/python3.5/si
pro 20 18:40:47 raspberrypi python[19188]:     self._detect_block()
pro 20 18:40:47 raspberrypi python[19188]:   File "/home/pi/env/lib/python3.5/si
pro 20 18:40:47 raspberrypi python[19188]:     raise LangDetectException(ErrorCo
pro 20 18:40:47 raspberrypi python[19188]: langdetect.lang_detect_exception.Lang

```

shivasiddharth, 2018-12-20:
What is the name of the radio that you have assigned ?

SkyCZ, 2018-12-21:
> 
> 
> What is the name of the radio that you have assigned?

If you mean stationnames: its "radio 1"
if you mean stationradio: its "Evropa 2"

"Evropa 2" is the official public name of the radio too

shivasiddharth, 2018-12-22:
Fixed the radio thingy. Replace your existing actions.py with the updated file from master. For youtube can you be a bit clear. Is it not playing thew first song only or is it playing only the first song? Also can you give the syntax that you used for the youtube command.

SkyCZ, 2018-12-23:
its playing only the first song... And i used "Hey google, Play imagine dragons playlist on youtube" (for example) it said getting youtube link... played the 1st song and then it was quiet...

SkyCZ, 2018-12-23:
And with the new actions.py im getting

```
pro 23 16:56:15 raspberrypi python[8566]:   {"text": "play Radio 1"}
pro 23 16:56:15 raspberrypi python[8566]: http://icecast3.play.cz/evropa2-128.mp
pro 23 16:56:24 raspberrypi python[8566]: Tuning into Evropa 2
pro 23 16:56:26 raspberrypi python[8566]: High Performance MPEG 1.0/2.0/2.5 Audi
pro 23 16:56:26 raspberrypi python[8566]:         version 1.23.8; written and co
pro 23 16:56:26 raspberrypi python[8566]:         free software (LGPL) without a
pro 23 16:56:26 raspberrypi python[8566]: Directory: /tmp/
pro 23 16:56:26 raspberrypi python[8566]: Playing MPEG stream 1 of 1: female-say
pro 23 16:56:26 raspberrypi python[8566]: MPEG 2.0 L III cbr32 24000 mono
pro 23 16:56:28 raspberrypi python[8566]: [0:01] Decoding of female-say.mp3 fini
pro 23 16:56:28 raspberrypi sudo[8921]:       pi : TTY=unknown ; PWD=/home/pi ;
pro 23 16:56:28 raspberrypi sudo[8921]: pam_unix(sudo:session): session opened f
pro 23 16:56:28 raspberrypi sudo[8921]: pam_unix(sudo:session): session closed f
pro 23 16:56:28 raspberrypi sudo[8930]:       pi : TTY=unknown ; PWD=/home/pi ;
pro 23 16:56:28 raspberrypi sudo[8930]: pam_unix(sudo:session): session opened f
pro 23 16:56:28 raspberrypi sudo[8930]: pam_unix(sudo:session): session closed f
pro 23 16:56:28 raspberrypi python[8566]: [0368f2e8] vlcpulse audio output error
pro 23 16:56:28 raspberrypi python[8566]: ON_CONVERSATION_TURN_FINISHED:
pro 23 16:56:28 raspberrypi python[8566]:   {"with_follow_on_turn": false}
pro 23 16:56:28 raspberrypi python[8566]: ON_CONVERSATION_TURN_FINISHED:
pro 23 16:56:28 raspberrypi python[8566]:   {"with_follow_on_turn": false}
pro 23 16:56:29 raspberrypi python[8566]: [036918a8] prefetch stream error: unim
```

shivasiddharth, 2018-12-24:
If you take a look at the README, under the Youtube customization, in the syntax table, it is mentioned that, it pick a random song from the playlist and play it. https://github.com/shivasiddharth/GassistPi#adding-youtube-api-and-generating-api-key  So its working as it was designed to work, i dont see a problem.

For radio, can you share the link for the radio station that you are using ?

SkyCZ, 2018-12-24:
> 
> 
> If you take a look at the README, under the Youtube customization, in the syntax table, it is mentioned that, it will pick a random song from the playlist and play it. https://github.com/shivasiddharth/GassistPi#adding-youtube-api-and-generating-api-key So its working as it was designed to work, i dont see a problem.
> 
> For radio, can you share the link for the radio station that you are using ?

I thought it will play the whole playlist not just 1 random song from it

http://icecast3.play.cz/evropa2-128.mp3

shivasiddharth, 2018-12-24:
Can you share the full terminal errors for Radio? As you can see the errors are truncated. 
From what i see the Radio station is indeed playing. It can be slow because of network related issues or the volume of the vlc player might be down. 
To verify if the music volume is low , open the /home/pi/ directory, right click and choose to view hidden files after that, open the .mediavolume.json file. If you see a number like "15", try giving the command "Hey google, set music volume to maximum".  
If its a network related issue, it will start playing after loading the buffer, so give it a few minutes.

SkyCZ, 2018-12-24:
It was the music volume! Sorry for being that dumb!

SkyCZ, 2018-12-24:
is there any way of playing full playlists automatically for example from Spotify or youtube?

shivasiddharth, 2018-12-28:
With some minor tweaks to the existing codes, its possible to do that. As the primary issue has been resolved, closing this issue.

closed 2018-12-28 by shivasiddharth