shivasiddharth/GassistPi #694: OSMC Kodi NAS access issue

V1JCM, 2018-12-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000001e7d2e18

```
Paste your Board and OS details here.   
```

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
We have all our media in a NAS with Samba. I am able to command the OSMC Kodi via GassistPI, like go home or change volume, but accessing the media in the NAS hangs the Gassistpi. It says about accessing the files and then stops, like going on a neverending loop. I have this in config.yaml:
  musicdirectory: 'smb://seagate-41ddbb.local/public/Music/_KODI_/'
And GassistPI Raspbian is able to access the files just fine.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
play songs by Abba on Kodi
```

### Paste the contents of the terminal below
```

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MEDIA_STATE_IDLE
ON_MEDIA_STATE_IDLE
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by Abba on Kodi"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by Abba on Kodi"}
Searching for your file
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
Playing WAVE '/tmp/male-say.wav' : Signed 16 bit Little Endian, Rate 24000 Hz, Mono
ERROR:__main__:'result'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 728, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 593, in main
    kodiactions(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 729, in kodiactions
    singleplaykodi(phrase)
  File "/home/pi/GassistPi/src/actions.py", line 581, in singleplaykodi
    nummusicfiles=len(musicfiles["result"]["files"])
KeyError: 'result'
INFO:snowboy:Keyword 1 detected at time: 2018-12-21 18:18:33

AFTER THIS THE GASSISTPI STOPS REACTING TO ANYTHING, HAS TO STOP IT MANUALLY CNTRL-C
```

shivasiddharth, 2018-12-21:
You need to mount the NAS folder locally on Pi and then use that path in the config.yaml. This was discussed in one of the earlier issues, forgot the number though.

V1JCM, 2018-12-22:
No rock here... Any suggestions?

I mounted the Samba NAS as MyNAS, in config.yaml:
  musicdirectory: '/home/pi/MyNAS/Music/_KODI_/'

and the result:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by Abba on Kodi"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by Abba on Kodi"}
Searching for your file
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
Playing WAVE '/tmp/male-say.wav' : Signed 16 bit Little Endian, Rate 24000 Hz, Mono
ERROR:__main__:'result'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 728, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 593, in main
    kodiactions(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 729, in kodiactions
    singleplaykodi(phrase)
  File "/home/pi/GassistPi/src/actions.py", line 581, in singleplaykodi
    nummusicfiles=len(musicfiles["result"]["files"])
KeyError: 'result'
INFO:snowboy:Keyword 1 detected at time: 2018-12-22 17:54:46

shivasiddharth, 2018-12-22:
Please check #656

V1JCM, 2018-12-22:
Oh, ok! I too am running a separate Kodi, and a 41GB music collection...

shivasiddharth, 2019-01-19:
Do we have an update on this issue ?

V1JCM, 2019-01-19:
Sorry, I have been busy elsewhere (but tonight decided to give your new voice a try!).

I did a rescrape of Kodi music, everything seems to be in order kodiwise. But the Abba is not playing on GassistPI. Any advice?

shivasiddharth, 2019-01-20:
Follow https://github.com/shivasiddharth/GassistPi/issues/656#issuecomment-442395699

shivasiddharth, 2019-02-11:
Closing this for inactivity.

closed 2019-02-11 by shivasiddharth