shivasiddharth/GassistPi #698: Gassistpi control from Android phone

reydecopas, 2018-12-23:
Hi, 
I have installed Gassistpi in the same SBC with domoticz
Is it possible to control gassistpi but instead of control through the mic and the speaker... control it far away from an Android phone  with 'ok google'?

Sorry if this is not the right place for asking this kind of questions.

Great project!!!! Congrats!!!!

aandroide, 2018-12-27:
I stay tuned also interests me

shivasiddharth, 2018-12-28:
Closed using #703

closed 2018-12-28 by shivasiddharth

aandroide, 2018-12-28:
it's not really what the user asked, in my opinion.
definitely intended to make the phone assistant interact with the assistant on the raspberry by saying words that trigger the assistant even remotely.
Surely you can also do with mqtt, just be a little 'practical in developing a custom application on the phone that integrates the voice and send the strings to mqtt on the raspberry

shivasiddharth, 2018-12-28:
Frankly, I missed the part in the query with "ok google". But in the MQTT Dash app, users can create switch tiles that sends MQTT commands.

shivasiddharth, 2018-12-28:
@aandroide ,  You do not need a custom application. I have explained as to how it can be achieved in #703.

shivasiddharth, 2019-01-29:
@aandroide  Chekout this commit https://github.com/shivasiddharth/GassistPi/commit/ba899e9edbcc5835df3d4a95045484b7ae8c776b