shivasiddharth/GassistPi #700: Cannot play Google Music locally / silence

markusza, 2018-12-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Raspberry PI 3 - ARMv7 Processor rev 4 (v7l)
Hardware        : BCM2835
Revision        : a02082

Raspbian GNU/Linux 9 (stretch)   

Respeaker with 3.5mm speakers as audio
```

### Describe the bug:
```
Not sure if I'm doing something wrong. Clean Raspbian install with latest GassitPi. Authorized and refreshed google music. When I ask it to play something from Google music, it says "I can't do that here, but you can ask me to play it on one of your other devices."

Perhaps similar issue [#137](https://github.com/shivasiddharth/GassistPi/issues/137#issuecomment-395499779)
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
pi@hassio:~ $ ./env/bin/python -u /home/pi/GassistPi/src/main.py
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#configList', 'data': {'entries': [{'key': 'canManageFP', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'shouldShowLocalSuggestions', 'kind': 'sj#configEnt
ry', 'value': 'true'}, {'key': 'purchaseAccess', 'kind': 'sj#configEntry', 'value': '3'}, {'key': 'sideloadedPlaybackLogging', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isWarmWelcomeDisabledFo
rWoodstock', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'forceSignupFlow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'isNautilusUser', 'kind': 'sj#configEntry', 'value': 'true'}, {'key'
: 'isLNEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'canCancelSubscription', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'ytContent', 'kind': 'sj#configEntry', 'value': 'false'},
 {'key': 'nashvillebg', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'partridgeAccess', 'kind': 'sj#configEntry', 'value': '3'}, {'key': 'isVsAvailable', 'kind': 'sj#configEntry', 'value': 'false'
}, {'key': 'canBrowseSt', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'warmWelcomeButtonText', 'kind': 'sj#configEntry', 'value': 'LISTEN_NOW'}, {'key': 'appstoreRatingPrompt', 'kind': 'sj#configE
ntry', 'value': 'false'}, {'key': 'isTrAvailable', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isInAppMessageEnabled', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'disableInfoCardsOnDay
One', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'canRC', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'nuqRequiredForWsSignup', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'sear
chClusterResponseEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'enableWsExtras', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'accountType', 'kind': 'sj#configEntry', 'value': 'Unl
imited, Family'}, {'key': 'topListMaxSize', 'kind': 'sj#configEntry', 'value': '500'}, {'key': 'isSuggestRequestEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'showLMark', 'kind': 'sj#confi
gEntry', 'value': 'false'}, {'key': 'conciergeListenNow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'minimumAppStartCountForFreeUpsell', 'kind': 'sj#configEntry', 'value': '6'}, {'key': 'experim
entIdsToLog', 'kind': 'sj#configEntry', 'value': '12600100,12600335,12600357,12600366,12600388,12600390,12600394,12600397,12600426,12600428,12600430,12600446,12600465,12600466,12600467,12600506,12600526,1
2600530,12600533,12600536,12602068,12604244,12604245,12604246,12611729,12612188,12612247,12616614,12616615,12617467,12619023,12619033,12619127,12620830,12625361,12625500,12625941,12626008,12626945,1262795
1,12628654,12629142,12629667,12630065,12630209,12631091,12631414,12631643,12632064,12632720,12633320,12634000,12635213,12636020,12641836,12641837,12642809,12643537,12647853,12648818,12648954,12649047,1265
0240,12650284,12650285,12652170,12653224,12653393,12653567,12655808,12656072,12656556,12656567,12657925,12658555,12658847,12660587,12661698,12664007,12664317,12664580,12665507'}, {'key': 'isExplicitRestri
ctedByPControl', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isAnyServiceAvailable', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'pieDayOneCard', 'kind': 'sj#configEntry', 'value': 'NO_E
XPERIMENT'}, {'key': 'showYoutubeRedInfoCard', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isAcceptedUser', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'appStartCountForFreeUpsell', 'kin
d': 'sj#configEntry', 'value': '250'}, {'key': 'concListenNow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'wsAccess', 'kind': 'sj#configEntry', 'value': '1'}, {'key': 'instantSearchDelay', 'kind
': 'sj#configEntry', 'value': '300'}, {'key': 'isPodcastSyncEnabled', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'topListUpdatePeriodSeconds', 'kind': 'sj#configEntry', 'value': '604800'}, {'key
': 'isSuggestDeletionEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'androidTvEntityPageSize', 'kind': 'sj#configEntry', 'value': '10'}, {'key': 'shouldShowServerSuggestions', 'kind': 'sj#c
onfigEntry', 'value': 'true'}, {'key': 'situationsResponseTTLMinutes', 'kind': 'sj#configEntry', 'value': '30'}, {'key': 'isPlSharingEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'isFoples
sNautilusUser', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'showLNInSettings', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'appStartCountForFreeUpsellRevisionNumber', 'kind': 'sj#config
[... 126 lines cut ...]
### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Same as above
```

### Paste the contents of the terminal below
```
Same as above
```

shivasiddharth, 2018-12-27:
Your command is not in accordance with the syntax that I have given here. https://github.com/shivasiddharth/GassistPi/blob/master/README.md#music-streaming-from-google-music

markusza, 2018-12-27:
```
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all songs from artist Linkin Park from Google music"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all songs from artist Linkin Park from Google music"}
stopping vlc
linkin park
Looking for songs rendered by the artist
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
DEBUG:gmusicapi.Mobileclient1:GetStreamUrl(args=['0afbecf8-965e-32af-a13e-8623c7c3b52d', 'B827EB9BF15A', 'hi'], kwargs={'required_auth': [False, False, True, False]})
Segmentation fault

```

shivasiddharth, 2018-12-27:
In the /home/pi directory you will have one file called songs.json please attach the file here.

markusza, 2018-12-27:
[songs.json.log](https://github.com/shivasiddharth/GassistPi/files/2713354/songs.json.log)

shivasiddharth, 2018-12-27:
Its actually working. If you see, there are 28 instances of "Artist not found" and from the songs.json, i see that you have 12 songs by Linkin Park.
`DEBUG:gmusicapi.Mobileclient1:GetStreamUrl(args=['0afbecf8-123-32af-231-8623c7c3b52d', 'B827EB9123123', 'hi'], kwargs={'required_auth': [False, False, True, False]})`
means that, it has found a matching track and wants the url from the Google Music. The thing has crashed with an error due to issues with google cloud. 

Can you give it a shot again?

tmxkn1, 2019-01-12:
Also can't play google music. I don't subscribe to GMusic, but I have uploaded a lot of music to the library.

The contents in the console are:

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all music from Google Music"}

stopping vlc
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}`
```

System info:

```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000000932816f
```

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I don't have `songs.json`. 

I can ask it to stream music from Youtube.

I have also authorised GMusic.

shivasiddharth, 2019-01-16:
> ON_RECOGNIZING_SPEECH_FINISHED:
>   {"text": "play all music from Google Music"}

Your command syntax is wrong.

As the OP of the thread has not got back, closing this issue for inactivity.

closed 2019-01-16 by shivasiddharth