shivasiddharth/GassistPi #702: Playing music from a Spotify playlist with Gassistant

jcksnmdy, 2018-12-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000061002412

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
### Describe the bug:
```
I Messed up my last issue so......
[... 165 lines cut ...]
### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'xxxxxxxxxxxxxxxxxxxxxx'
```

### Paste the contents of the terminal below
```
same as above
```

shivasiddharth, 2018-12-28:
Segmentation fault is a cloud issue. Give it multiple shots and let me know if the error is occurring each and every-time or just once in a while.

jcksnmdy, 2018-12-28:
It is happening multiple times.

shivasiddharth, 2018-12-28:
Its a google cloud issue. Give it some time and try again.

shivasiddharth, 2018-12-29:
This has nothing to do with the scripts and i have no control over the error. I will leave this issue open. Can you run the following in the terminal and post the output.

```
source env/bin/activate
pip show youtube-dl
```

jcksnmdy, 2018-12-30:
~~~
Name: youtube-dl
Version: 2018.12.17
Summary: YouTube video downloader
Home-page: https://github.com/rg3/youtube-dl
Author: Ricardo Garcia
Author-email: ytdl@yt-dl.org
License: Unlicense
Location: /home/pi/env/lib/python3.5/site-packages
Requires: 
Required-by: 
~~~
What is OpenSSL

jcksnmdy, 2019-01-08:
I have done a new installation but i still get a segmentation fault with a different reason. 
```
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-01-08 03:56:42,823 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01b9fe80] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01bcce28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
[... cut ...]
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Segmentation fault
```

kasteelharry, 2019-01-08:
I am currently having the exact same issue,when I say "OK Google play test playlist from Spotify". It says getting music links and then gives the error that was first placed in this issue ending with segmentation fault.

aandroide, 2019-01-08:
@kasteelharry did you correctly enter your spotify credentials in config.yaml?

jcksnmdy, 2019-01-08:
Is there a way to fix it on our end

aandroide, 2019-01-08:
@jcksnmdy we have to wait for the google team to set up the sdk library.
for your case did you follow my advice?

shivasiddharth, 2019-01-09:
@jcksnmdy Can you check if you have entered the Google API credentials correctly. Your spotify part is fine, because its getting the playlist. Its crashing only when trying to fetch the links from youtube. The API credentials should be entered within "" double quotes.

jcksnmdy, 2019-01-09:
@aandroide I have check three times my credintials are fine
@shivasiddarth What exactly are you talking about, device model id, project id, model id. Or something else??

shivasiddharth, 2019-01-09:
https://github.com/shivasiddharth/GassistPi/blob/2587341cee46c66836f5872588c09ee442c5e903/src/config.yaml#L76-L77

jcksnmdy, 2019-01-09:
Ooooo I have not done that.... thanks so much. I will get back to you to you in about 3 hours if it works.

kasteelharry, 2019-01-09:
@aandroide, I discovered my mistake. I had not setup my google api in the config.yaml yet and after setting up the key I was able to stream my spotify playlists.

shivasiddharth, 2019-01-10:
Closing this issue then.

closed 2019-01-10 by shivasiddharth