shivasiddharth/GassistPi #708: ON_MEDIA_STATE_IDLE

dsvavishal2003, 2018-12-30:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a21041
Serial		: 00000000df98f327
--------------------------------------------------------------------------------------------------------------------------------------

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
### Describe the bug:

When I run the gassist-installer.sh script, it normally installs the python modules and all the other stuff and prompts me to authorise the assistant. When I authorise the assistant with my Google account, it installs the assistant and starts the demo assistant automatically. But then, the assistant automatically mutes and doesn't respond to any commands or to the 'OK-Google' hotword.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Registering...Done.

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_MEDIA_STATE_IDLE

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Well, the demo assistant just started, but didn't respond as mentioned earlier

### 4. Is the assistant service restarting automatically?
I don't know. Actually, I haven't properly started it once in the first place.

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

dsvavishal2003, 2018-12-30:
Now, i installed the services for auto start on bootup. And then rebooted it. Now, the Pi is not at all booting up. Both the LED's stay ON without blinking and I'm not getting a display via HDMI. I googled a bit and found out at Stack Overflow that this is because the image in the SD card has become corrupt.
Now, what should I do, Sid...?

shivasiddharth, 2018-12-30:
Mate, this issue forum is only for troubleshooting any issues pertaining to this project. I would suggest you to use additional resources or references for issues outside the scope of this work.