shivasiddharth/GassistPi #709: Speaker and Microphone works fine, Responds to "Hey Google", but does not listen to basic commands.

lavos3rd, 2018-12-31:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial	: 00000000f42deff6 

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
Followed the instructions to the T.
Tested my microphone just using arecord, played it back with aplay. Audio is clear and no static.
I manually started the Assistant, few seconds later, Assistant said: "Hi" (an indication she's ready). I then said "Hey Google...(a tone could be heard) ...Set volume to maximum". Minutes passed, no response, and then she either said "theres something wrong" or "theres a glitch".

[... 110 lines cut ...]
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_RESPONDING_STARTED:
  {"is_error_response": true}

ON_RESPONDING_FINISHED

ON_ASSISTANT_ERROR
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

lavos3rd, 2018-12-31:
demo test query? which part in the steps is that? Im sure I've enabled the assistant API

shivasiddharth, 2018-12-31:
Demo of the assistant will start automatically after installation just after authorization.

lavos3rd, 2019-01-01:
Ah yes, The demo test... it's the same issue when the authorization finishes.

shivasiddharth, 2019-01-01:
If even the demo did not go through, its clearly an assistant api issue. Are you in a region unsupported by google assistant ?

lavos3rd, 2019-01-02:
Oh, I'll check, but I don't think Philippines is in the blacklist

kennethgomad02, 2019-01-02:
Hey their, I believe that you have some error on output sound 'jack server is not running or cannot be started' so you need to do is install pulse by using 'sudo apt-get install pulseaudio' and when its done quick do 'sudo nano /etc/pulse/daemon.conf ' this command enable all channel for soundcard and uncomment default-sample-channels (ie remove the semicolon at the beginning of the line) and set it to 6 if you System 5.1 or 8, if your system is 7.1. If your using Raspbian Stretch Dekstop change the 'default-sample-channels = 2 to default-sample-channels = 6' make sure to save the file control x and press enter. Once your done do 'pulseaudio --kill' if its says about daemon error message then its good and finaly run the pulseaudio by entering 'pulseaudio --start' .   And to be sure do sudo raspi-config and force the sound output to port jack shown below. And say "Hey Google set volume to maximum".
[image]

[image]

[image]

[image]

-

kennethgomad02, 2019-01-02:
Also if your using USB mic then you to set it also in alsa mixer, and do please check sudo cd ~ && nano .asoundrc if the file is empty you need to change it as the command you use from the instruction doesn't work . And enter this into that file this is originally installed from Alexa but this should work! Tell me if it's works.
[image]

shivasiddharth, 2019-01-02:
Are you behind a firewall or vpn ?

lavos3rd, 2019-01-04:
Nope, No VPN and No Firewall

shivasiddharth, 2019-01-04:
Will leave this open to see if someone else reports the same.

lavos3rd, 2019-01-04:
Thanks, I'll try and do it again from the start

chiragatha, 2019-01-22:
@lavos3rd @shivasiddharth did you guys got the solution for this issue. Even I have got the same problem #761

shivasiddharth, 2019-02-12:
Closing this for inactivity.

closed 2019-02-12 by shivasiddharth

akshathjkxx007, 2019-06-29:
I am also getting an error response like there is a glitch.But for sometime it worked fine and then when I ask something it says that something went wrong please try again in a few seconds.
Plz help urgent