shivasiddharth/GassistPi #714: kodi install

remym777, 2019-01-04:
Hi, thanks to you i have been able to install Google Assistant and it works great.
I’m now trying to follow the vidéo tutorial to enable the kodi command but i encounter a problem.

On your vidéo tutorial the action.py look like this...
[URL=https://www.casimages.com/i/190104060046243044.png.html][IMG]https://nsa39.casimages.com/img/2019/01/04/190104060046243044.png[/IMG][/URL]

Mine look like this. Many lines about youtube and kodi are missing. I have nowhere to put the API Numbers like you say.
[URL=https://www.casimages.com/i/190104055727498982.png.html][IMG]https://nsa39.casimages.com/img/2019/01/04/190104055727498982.png[/IMG][/URL]

I must have missed something but after looking everywhere i can’‘t find WHAT..

One last thing. Since your tutorial the Google webpage changed. I can’t find like in your tutorial where to tick In blue the API. SEE PICTURE below 
[URL=https://www.casimages.com/i/190104060635623822.png.html][IMG]https://nsa39.casimages.com/img/2019/01/04/190104060635623822.png[/IMG][/URL]

Thanks a lot by advance.

shivasiddharth, 2019-01-04:
The videos are only for reference. README of the project is the BIBLE. The details have to be entered in the config.yaml and nowhere else.

closed 2019-01-04 by shivasiddharth

remym777, 2019-01-05:
Hi and thanks for your answer. on the readme file on step 5 of the integration of kodi you write :Make a note of the created API Key and enter it in the actions.py script at the indicated location.
but i can't find the indicated location either on the action.py or looke you said previously ont your mail on the config.yampl

thanks by advance

shivasiddharth, 2019-01-05:
Like i mentioned before, 

> The details have to be entered in the config.yaml and nowhere else.
The mention about actions.py in step 5 is a typo and i have fixed it now. In the config.yaml, the API key needs to be entered in this line.
https://github.com/shivasiddharth/GassistPi/blob/fc2b67f26b830332e4e4f30c0c14cfb587ce1e6d/src/config.yaml#L77