shivasiddharth/GassistPi #719: Assistant crashes with segmentation fault when asked for ingredients

jcksnmdy, 2019-01-07:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000061002412

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
### Describe the bug:
```
Assistant gives the ingredients and then crashes. 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes, I know that it just pushes the ingredients to your phone and does not say the ingredients but it also crashes with segmentation fault.

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'XXXXXXXXXXXXXXXXXXXXXXXXX'

      """""""""Get ingredients for squash pudding""""""""""

```

### Paste the contents of the terminal below
```
        ***************startup is normal***********************
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "get ingredients for squash pudding"}

https://api.edamam.com/search?q=squash%20pudding&app_id=3976739d&app_key=3cafc0b83bb60f3d7ed15eb79dfa6df2
Summer Squash Pudding

http://www.food52.com/recipes/5370_summer_squash_pudding

'4 yellow summer squash', '2 eggs, beaten', '1 tablespoon sugar', '1 tablespoon butter', '2 tablespoons sour cream', 'pinches salt and pepper', '1 cup breadcrumbs'
Segmentation fault

```

jcksnmdy, 2019-01-07:
Thank you will do!!!

jcksnmdy, 2019-01-08:
I have redone the installation and i still get segmentation fault after it gives me the ingredients.

aandroide, 2019-01-08:
have you connected the pushbullet account too?
otherwise disable line 1278
 # pushmessage ( str (recipe_name), str (compiled_recipe_info))

the segmentation error is caused by the last sdk update that has nothing to do with gassistpi, this error has already been reported to the google assistant sdk team.
the error is shown when we are thrown out or simply by stopping the assistant by pressing ctrl + x
in your case I think the error is shown to you because you have not inserted the pushbullet api key and you are thrown out.

jcksnmdy, 2019-01-08:
Thanks so much it works without error

shivasiddharth, 2019-01-09:
@aandroide  Thanks for stopping by. I missed seeing that.. Closing this issue.

closed 2019-01-09 by shivasiddharth

aandroide, 2019-01-09:
Do not worry, it's a pleasure