shivasiddharth/GassistPi #722: Assistant not responding to custom wakewords & mic is automatically going to Mute mode

dsvavishal2003, 2019-01-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a21041
Serial		: 00000000df98f327

-------------------------------------------------------------------------------------------------------------------------------------------
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:

   I have installed GassistPi in my Raspberry Pi 2. Before tweaking around with the code for the first time, the assistant was working perfectly. Then, to enable custom wakeword activation, I set the Ok_Google parameter in the config.yaml file to 'Disabled' and set the custom wakeword to 'Enabled'
Here's how the code looked like after it was finished 

```
#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Disabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    -'/home/pi/GassistPi/src/resources/sanjo.pmdl'
```

Then, I restarted the assistant. At first, it booted up normally. At that time, I called out Alexa. It didn't respond. Then, I tried 'OK Google'. Immediately, it responded. Then, it said, "Mic is muted". After that, it just doesn't respond to anything. I tried searching the issue page and found out that we had to delete the .mute file. But, the simple problem is, there is no .mute file. 
So, the issues here are 
1. GassistPi automatically goes to mute mode after booting up.
2. The 'OK Google' wakeword still works after setting the parameter to 'Disabled' in the config.yaml file

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
Sorry, there is no terminal.
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2019-01-10:
> GassistPi automatically goes to mute mode after booting up.
> The 'OK Google' wakeword still works after setting the parameter to 'Disabled' in the config.yaml file

If you set the Ok_Google to disabled, then mic will be muted, but will still to Custom wakewords. There is no way that it will listen to Ok_Google once its muted. It will listen to Ok_Google only when the mic is open. From what you are saying, there is only one thing that could have happened. Your Alexa would have opened the Mic and hence while the mic was listening to your input, your Ok_Google would have been picked up.

dsvavishal2003, 2019-01-10:
But @shivasiddharth, this is what is happening. 
It could be a bug. I completely reinstalled it. But still, it is showing the same issue. Any ideas..? I trained my own custom voice model using Snowboy. And then tried using it too. But still, to no gain.

shivasiddharth, 2019-01-10:
Stop the service, manually run the assistant, perform the same steps what you did above and share the terminal contents here.

dsvavishal2003, 2019-01-10:
How do we stop the assistant ..?

shivasiddharth, 2019-01-10:
`sudo systemctl stop gassistpi.service`

dsvavishal2003, 2019-01-13:
I stopped and started the assistant manually. And hurrah..!! It's listening to all the custom wakewords.. But, I the only remaining problem is that it says mic is muted. When we call Sanjo, it says mic is back on. Then only does it respond to wakewords. The contents of the terminal are : 
`
ON_MUTED_CHANGED:
  {"is_muted": true}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:05
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:11
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:13
ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the time now"}

ON_RENDER_RESPONSE:
  {"text": "It's 8:15.", "type": 0}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:30
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:36
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:15:39
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}

ON_CONVERSATION_TURN_TIMEOUT

ON_MUTED_CHANGED:
  {"is_muted": true}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:16:18
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:16:20
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who is the first prime minister of India"}

ON_RENDER_RESPONSE:
  {"text": "Jawaharlal Nehru was the 1st Prime Minister of India.", "type": 0}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:16:32
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:16:34
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who do you how are you doing"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_RENDER_RESPONSE:
{
  "text": "I'm doing great, thanks for asking. What can I help you with?",
  "type": 0
}

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}

ON_CONVERSATION_TURN_STARTED

INFO:snowboy:Keyword 1 detected at time: 2019-01-13 08:16:51
ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what are my events for today"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

`
Please do respond ..

shivasiddharth, 2019-01-13:
> When we call the custom wakeword, it says mic is back on. Then only does it respond to wakewords Then , we have to again call the wakeword to make it respond. Any ideas why this is happening?

When you say Mic is back on you mean an audio out from the assistant? This is impossible. I have added the voice prompts for Mic-On/Off only for GPIO based muting.

dsvavishal2003, 2019-01-13:
Yes I do mean that. The assistant says it. Mic is Back On. Yes yes yes... Then if we again call the wakeword, it plays that little sound indicating that the assistant is listening. Then, if we speak something, it responds...

shivasiddharth, 2019-01-13:
Yes indeed... It seems to be an issue with Google's Latest SDK that they are adding voice over prompts from the cloud. Was not the case before. Adding a bug tag. Will fix it shortly.

closed 2019-01-13 by shivasiddharth

shivasiddharth, 2019-01-13:
A have added a number of commits since the last time you may have cloned the git. If have to update your project folder with the latest scripts from the git. Updating the project has already been discussed in gitter quite a few times.

dsvavishal2003, 2019-01-13:
I'm sorry, but could you please leave a link to those discussions...?

shivasiddharth, 2019-01-13:
https://gitter.im/GassistPi/Lobby/ The links are there in the readme in bold already.

dsvavishal2003, 2019-01-17:
Also, @shivasiddharth, is there any way to change the voice of the assistant to male...?
I changed it in the actions console, but to no gain... Is there any way...?

skrajeev, 2019-05-31:
The same problem is still there, gassistPi responds to custom word, after responding to the question asked it says " the mic's muted". And when I trigger it using custom wakeword it says "the mic's back on".

Nivin-Abraham, 2020-06-14:
@shivasiddharth Budyy first of all hats off to the Gassitpi, you rock made my life very easy to set this up and kudos to that.

I still face the issue after setting the custom wakeword, every time i use the custom wakeword, it says "mic is back" and then it says "mic is muted", specialy when i use it to play songs from gaana.com, it repeats the same every 2or 3 seconds and interrupts the song, could you kindly have a check on it.

commit 7d1245fa (2019-01-13): Fix Ok-Google-Disabled and Custom wakeword not working issue

Fixes #722
commit 643efccf (2019-01-13): Update main.py

Fixes #722