shivasiddharth/GassistPi #723: MQTT

DewGew, 2019-01-08:
I Need to send a mqtt message somewhere in main.py or actions.py to my domoticz PI to change a selector switch when I start my radio.
sending switch command to domoticz:
`{"command": "switchlight", "idx": 123, "switchcmd": "Set Level", "level": 10 }`
How can I manage to do this?

shivasiddharth, 2019-01-08:
In the main.py i have already defined the client. Convert the json to string using json.dumps. then use client.publish(mqtt topic, mqtt message).

Please use gitter for non-issue stuff.

closed 2019-01-08 by shivasiddharth

NGOLOC37, 2022-08-08:
I use client.publish but got error. Can u help me. Thanks!
[image]