shivasiddharth/GassistPi #724: [Feature Request] Repeat after me

DewGew, 2019-01-08:
A function like " Repaet after me" would be useful.
If I say " Repeat after me, Hi I'm new to google assistant ":
Then GassistPi replays: " Hi I'm new to google assistant "

With this I can send out custom tts to my GassistPi from my domoticz via mqtt.

DewGew, 2019-01-08:
Repeat after me is a native function already but I cant make this command with mqtt.

DewGew, 2019-01-09:
I just need a function and send custom voice notifications true gassistpi. Should be possible to make a custom function for this.

DewGew, 2019-01-10:
I created my own function. I added this:
in **actions.py**:
```
def speak(phrase):
    voice_notify = phrase.replace("speak", "")
    say(voice_notify)
```
and in **main.py**:
`from actions import speak`
```
if 'speak'.lower() in str(usrcmd).lower():
    self.assistant.stop_conversation()
    pippa(str(usrcmd).lower())
```
with this I can send voice notifications with mqtt command:

`custom speak I am new to google assistant`

and Gassist Pi replies "Im new to google assistant"

Now only thing is that swedish characters not working via mqtt. I can't send å ä and ö.
I only get those like `'\xc3\x85', '\xc3\x84', '\xc3\x96'. Any ideas howto get this to work?
`

jcksnmdy, 2019-01-12:
I get segmentation error when running this in assistant. What might I have done wrong?

aandroide, 2019-01-12:
Now only thing is that swedish characters not working via mqtt. I can't send å ä and ö.
I only get those like '\xc3\x85', '\xc3\x84', '\xc3\x96'. Any ideas howto get this to work?

the problem is in action.py in the "word translator" function is not declared unicode utf-8
#Word translator
def trans(words,destlang,srclang):
    transword= translator.translate(words, dest=destlang, src=srclang)
    transword=transword.text
    transword=transword.replace("Text, ",'',1)
    transword=transword.strip()
    print(transword)
    return transword

DewGew, 2019-01-12:
Word translate is not the problem even print() does not work correct.

aandroide, 2019-01-12:
your print is wrong you have to set it up and you'll see that it works:
print("Message from MQTT: "+str(msg.payload.decode("utf-8")))

aandroide, 2019-01-12:
if I write in Italian custom speak "è vero" the message is printed correctly but pronounced with the errors x
Message from MQTT: custom speak è vero
\ xc3 \ xa8 vero '
Playing WAVE '/tmp/male-say.wav' : Signed 16 bit Little Endian, Rate 24000 Hz, Mono

shivasiddharth, 2019-01-12:
Did you guys try adding the UTF-8 encoding at the top of the script ?

aandroide, 2019-01-12:
yes, don't work

shivasiddharth, 2019-01-12:
above this line 
https://github.com/shivasiddharth/GassistPi/blob/b737c460f86b80da059df573796514db00212ad7/src/actions.py#L315
add
`print(sayword)`
And above this line 
https://github.com/shivasiddharth/GassistPi/blob/b737c460f86b80da059df573796514db00212ad7/src/actions.py#L307
add
`print(words)`

Now give the command and post as to what those lines are printing.

aandroide, 2019-01-12:
Message from MQTT: custom speak è vero 
 \xc3\xa8 vero '
\ xc3 \ xa8 vero '
Playing WAVE '/tmp/male-say.wav' : Signed 16 bit Little Endian, Rate 24000 Hz, Mono

aandroide, 2019-01-12:
adding that line hangs and writes:
Message from MQTT: <built-in method decode of bytes object at 0x6f43bc68>

aandroide, 2019-01-12:
mqtt does his job well and prints the accents correctly. the problem is something else

print ("Messaggio da MQTT:" + str (msg.payload.decode ("utf-8") ))

shivasiddharth, 2019-01-12:
Guess you have added it at wrong place. Adding `mqtt_query = mqtt_query.decode("utf-8")` after `mqtt_query=str(msg.payload)[1:] ` will not affect the `print("Message from MQTT: "+str(msg.payload.decode("utf-8")))`

aandroide, 2019-01-12:
setting it up so it locks completely:
def on_message(self, client, userdata, msg):
        if self.can_start_conversation == True:
            print("Message from MQTT: "+str(msg.payload.decode("utf-8")))
            mqtt_query=str(msg.payload)[1:]
            mqtt_query = mqtt_query.decode("utf-8")
            if 'custom' in mqtt_query.lower():
                mqtt_query=mqtt_query.lower()
                mqtt_queryidx=mqtt_query.find('custom')
                mqtt_query=mqtt_query[mqtt_queryidx:]
                mqtt_query=mqtt_query.replace('custom',"",1)
                mqtt_query=mqtt_query.strip()
                self.custom_command(mqtt_query)
            else:
                self.assistant.send_text_query(mqtt_query)

closed 2019-01-12 by shivasiddharth

aandroide, 2019-01-12:
good job @shivasiddharth

DewGew, 2019-01-14:
This does not work for me...

DewGew, 2019-01-14:
This does not work for me..

shivasiddharth, 2019-01-14:
Please look at the commit referenced above.

DewGew, 2019-01-15:
When I use Google cloud. It speak fast. Can I add "speakingRate" here somewhere?
https://github.com/shivasiddharth/GassistPi/blob/a2cbf57ca0c40c43c17a0b4b5e2f393c5a38fc70/src/actions.py#L279

commit 29d4d891 (2019-01-12): Fix unicode characters MQTT issue

Fixes #724