shivasiddharth/GassistPi #725: no module named paho

ykr412, 2019-01-09:
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id '*******'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 44, in <module>
    import paho.mqtt.client as mqtt
ImportError: No module named 'paho'

jcksnmdy, 2019-01-09:
‘’’
pip install paho-mqtt

cd Gassistant/src/
ln -s /home/pi/.local/lib/python2.7/site-packages/paho paho
ln -s /home/pi/.local/lib/python2.7/site-packages/paho
‘’’

Worked for me

shivasiddharth, 2019-01-10:
It means you are using updated scripts with old installations. Please update your project.

closed 2019-01-10 by shivasiddharth

ykr412, 2019-01-11:
what do you mean by updating the project, and how can i do that ?

shivasiddharth, 2019-01-11:
Check out the giiter lobby.

ykr412, 2019-01-11:
I have updated everything works fine but nothing in the config.yaml works , custom converstaion or youtube .. it crashes after i ask

ykr412, 2019-01-11:
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}

Who do you think I am
Segmentation fault
pi@raspberrypi:~ $