shivasiddharth/GassistPi #726: Bug to play Spotify and Deezer

vmagalhaes, 2019-01-10:
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
I`m trying to play music with Spotify or Deezer, but any of those are working

DEEZER
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Rock In Rio last day from Deezer"}

stopping vlc
/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning)
No matching playlists found
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Segmentation fault
Segmentation fault

SPOTIFY
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Rock In Rio last day from Spotify"}

stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
rock in rio last day
18
Segmentation fault

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

shivasiddharth, 2019-01-10:
Duplicate of #702

closed 2019-01-10 by shivasiddharth

vmagalhaes, 2019-01-10:
My config.yaml have Google Cloud API Configured :/

shivasiddharth, 2019-01-10:
Is it within quotes ? Also double check if your spotify credentials are correct, these are the only two things that could have gone wrong. Your post of the terminal is not full, its segmented in places. Means like you have copy pasted different parts at different points of time.

vmagalhaes, 2019-01-10:
Oh! I do not think it's in quotes, I'll change it and I'll return the result

vmagalhaes, 2019-01-11:
I add quotes in api and still bugging

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I hurt my self today from Spotify"}

stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Segmentation fault
i hurt my self today
18
Segmentation fault

```

#API Key for YouTube and Kickstarter Search Engine
Google_cloud_api_key: 'AIzaSyD07xK0ra3_qZULChq7XQoT9zDtCYcXxjw'

#Spotify credentials. Register for a spotify developer and get the credentials.
Spotify:
  Spotify_Control: 'Enabled'
  client_id: 'e9be7da8e5ff4079b2871e27eab94xxx' (x to hide here on github)
  client_secret: 'ad8f28469a2741e6b4c6605e41786xxx' (x to hide here on github)
  username: 'viictorg1'

Its all right? Do you know what can be?