shivasiddharth/GassistPi #728: setup.py bdist_wheel for lxml

divijbajaj5, 2019-01-10:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 00000000502e9230

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
``` It is getting hanged while building a wheel for lxml

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
Yes
### 2. Is the issue related to assistant installation ?
Yes/No
Yes
### If answer to question 2 is Yes, then paste the contents of the terminal below.

```Building wheels for collected packages: lxml
  Running setup.py bdist_wheel for lxml ... |

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

jcksnmdy, 2019-01-10:
I don’t know exactly what your asking but for me it just took a while for it to finally go thru. BE PATIENT!! Mine shut down my VNC server for a while so I let it sit. I came back a bout an hour later and my VNC window was back open and installation was done. If it still is taking a while I suggest restarting the installation and wait again.

divijbajaj5, 2019-01-10:
In my case, it is not responding after this command...
Running setup.py bdist_wheel for lxml ... |

I waited for hours to get it done but it is not responding. What should I do install the complete OS again?

jcksnmdy, 2019-01-10:
No just the gassistant installer

shivasiddharth, 2019-01-10:
Unless and until something crashes, it ain't a bug. This lxml thingy has already been discussed in gitter.

closed 2019-01-10 by shivasiddharth

m4ri064, 2019-01-26:
I get this error while instalation, 

```
  error: [Errno 28] No space left on device
  adding 'grpc/_cython/__init__.py'
  adding 'grpcio-1.18.0.dist-info/RECORD'
  error: [Errno 28] No space left on device  
  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Successfully built spotipy googletrans

cc tmp/xmlXPathInitq7kaouwu.o -lxml2 -o a.out
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
    
    ----------------------------------------
Command "/home/mario/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-bk3jqfgj/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-nj8epfw_/install-record.txt --single-version-externally-managed --compile --install-headers /home/mario/env/include/site/python3.5/lxml" failed with error code 1 in /tmp/pip-install-bk3jqfgj/lxml/
```