shivasiddharth/GassistPi #734: Download issue

aburaihanshawon, 2019-01-12:
When i execute the last command and enter my project id,model id it starts to download some file.at this time in a certain step it stop downloading forever and never goes to next step to download or install file.the last line like this "Running setup.py bdist_wheel for lxml...." what is the solution.

shivasiddharth, 2019-01-12:
Duplicate of #728

closed 2019-01-12 by shivasiddharth

piyushtanksale00, 2019-02-03:
When i run update.sh script it is showing project already up to date but files are not updated. config.yaml file is not showing update of  feb 3 2019

shivasiddharth, 2019-02-03:
Not sure when you cloned the git. get the latest update script from master.

shivasiddharth, 2019-02-03:
I have checked the update script multiple times and it works. Make sure to use them properly. Pull the latest update script and replace the old one that you have got. It has to be in the /scripts directory.