shivasiddharth/GassistPi #739: OSMC AUTOSTART does not work

fraraf85, 2019-01-15:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000eaf01f7c
osmc@osmc:~$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
osmc@osmc:~$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000eaf01f7c

PRETTY_NAME="Open Source Media Center"
NAME="OSMC"
VERSION="December 2018"
VERSION_ID="2018.12-1"
ID=osmc
ID_LIKE=debian

ANSI_COLOR="1;31"
HOME_URL="https://www.osmc.tv"
SUPPORT_URL="https://www.osmc.tv"
BUG_REPORT_URL="https://www.osmc.tv"

### Describe the bug:
```
I followed the assistant autostart guide, but it does not work
gassistpi.service is installed correctly
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
/No

### 4. Is the assistant service restarting automatically?
/No

osmc@osmc:~$ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-01-15 15:23:20 CET; 6s ago
 Main PID: 2833 (python)
   CGroup: /system.slice/gassistpi.service
           └─2833 /home/osmc/env/bin/python -u /home/osmc/GassistPi/src/main.py --device_model_id rasp-4959b-raspgoogle-9jozmf

gen 15 15:23:20 osmc systemd[1]: Started Google Assistant-Voice.

shivasiddharth, 2019-01-15:
> osmc@osmc:~$ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
Active: active (running) since Tue 2019-01-15 15:23:20 CET; 6s ago
Main PID: 2833 (python)
CGroup: /system.slice/gassistpi.service
└─2833 /home/osmc/env/bin/python -u /home/osmc/GassistPi/src/main.py --device_model_id rasp-4959b-raspgoogle-9jozmf
gen 15 15:23:20 osmc systemd[1]: Started Google Assistant-Voice.

Your assistant is running. If the wakeword is not being detected, check your mic.

closed 2019-01-15 by shivasiddharth

fraraf85, 2019-01-17:
it's not a microphone problem, because if I start manually the assistant works

aandroide, 2019-01-17:
try with: sudo systemctl restart gassistpi.service

shivasiddharth, 2019-01-17:
I suspect this to be that html issue. Run
```
sudo systemctl stop gassistpi.service
sudo systemctl disable gassistpi.service
```
open the service file within the systemd folder inside GassistPi folder, add a 0 or 1 at the end of your modelid. Re run the service installer.

fraraf85, 2019-01-18:
I tried how you wrote me but it does not work the same, I am attaching the change I made. I tried both to put 0 and 1

`[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/osmc/env
Environment=PATH=/home/osmc/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/osmc/env/bin/python -u /home/osmc/GassistPi/src/main.py --device_model_id 'xxxxxxxxxx' 0

WorkingDirectory=/home/osmc/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=osmc

[Install]
WantedBy=multi-user.target`

shivasiddharth, 2019-01-19:
not this way it should be 'xxxxxxxxx0'

fraraf85, 2019-01-20:
nothing even so does not function 'xxxxxxx0' and 'xxxxxx1'

fraraf85, 2019-01-21:
`osmc@osmc:~$ sudo journalctl -u gassistpi.service
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
gen 21 12:43:40 osmc python[291]: /home/osmc/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /homgen 21 12:43:40 osmc python[291]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
gen 21 12:43:40 osmc python[291]: 2019-01-21 12:43:40,601 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credegen 21 12:43:40 osmc python[291]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/osmc/.local/share/gmusgen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:failed to authenticate
gen 21 12:43:42 osmc python[291]: Traceback (most recent call last):
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/main.py", line 45, in <module>
gen 21 12:43:42 osmc python[291]:     from actions import say
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/actions.py", line 89, in <module>
gen 21 12:43:42 osmc python[291]:     custom_action_keyword = yaml.load(conf)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/__init__.py", line 70, in load
gen 21 12:43:42 osmc python[291]:     loader = Loader(stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/loader.py", line 34, in __init__
gen 21 12:43:42 osmc python[291]:     Reader.__init__(self, stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 85, in __init__
gen 21 12:43:42 osmc python[291]:     self.determine_encoding()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 124, in determine_encoding
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
gen 21 12:43:40 osmc python[291]: /home/osmc/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/gen 21 12:43:40 osmc python[291]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
gen 21 12:43:40 osmc python[291]: 2019-01-21 12:43:40,601 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentgen 21 12:43:40 osmc python[291]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/osmc/.local/share/gmusicgen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:failed to authenticate
gen 21 12:43:42 osmc python[291]: Traceback (most recent call last):
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/main.py", line 45, in <module>
gen 21 12:43:42 osmc python[291]:     from actions import say
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/actions.py", line 89, in <module>
gen 21 12:43:42 osmc python[291]:     custom_action_keyword = yaml.load(conf)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/__init__.py", line 70, in load
gen 21 12:43:42 osmc python[291]:     loader = Loader(stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/loader.py", line 34, in __init__
gen 21 12:43:42 osmc python[291]:     Reader.__init__(self, stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 85, in __init__
gen 21 12:43:42 osmc python[291]:     self.determine_encoding()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 124, in determine_encoding
gen 21 12:43:42 osmc python[291]:     self.update_raw()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 178, in update_raw
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
[... cut ...]
gen 21 12:43:43 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:logged out`

shivasiddharth, 2019-01-21:
The formatting is totally off. Can you paste that to a file and attach the file with formating in order?

fraraf85, 2019-01-21:
osmc@osmc:~$ sudo journalctl -u gassistpi.service
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
gen 21 12:43:40 osmc python[291]: /home/osmc/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /homgen 21 12:43:40 osmc python[291]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
gen 21 12:43:40 osmc python[291]: 2019-01-21 12:43:40,601 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credegen 21 12:43:40 osmc python[291]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/osmc/.local/share/gmusgen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:failed to authenticate
gen 21 12:43:42 osmc python[291]: Traceback (most recent call last):
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/main.py", line 45, in <module>
gen 21 12:43:42 osmc python[291]:     from actions import say
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/actions.py", line 89, in <module>
gen 21 12:43:42 osmc python[291]:     custom_action_keyword = yaml.load(conf)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/__init__.py", line 70, in load
gen 21 12:43:42 osmc python[291]:     loader = Loader(stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/loader.py", line 34, in __init__
gen 21 12:43:42 osmc python[291]:     Reader.__init__(self, stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 85, in __init__
gen 21 12:43:42 osmc python[291]:     self.determine_encoding()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 124, in determine_encoding
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
gen 21 12:43:40 osmc python[291]: /home/osmc/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/gen 21 12:43:40 osmc python[291]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
gen 21 12:43:40 osmc python[291]: 2019-01-21 12:43:40,601 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentgen 21 12:43:40 osmc python[291]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/osmc/.local/share/gmusicgen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:failed to authenticate
gen 21 12:43:42 osmc python[291]: Traceback (most recent call last):
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/main.py", line 45, in <module>
gen 21 12:43:42 osmc python[291]:     from actions import say
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/GassistPi/src/actions.py", line 89, in <module>
gen 21 12:43:42 osmc python[291]:     custom_action_keyword = yaml.load(conf)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/__init__.py", line 70, in load
gen 21 12:43:42 osmc python[291]:     loader = Loader(stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/loader.py", line 34, in __init__
gen 21 12:43:42 osmc python[291]:     Reader.__init__(self, stream)
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 85, in __init__
gen 21 12:43:42 osmc python[291]:     self.determine_encoding()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 124, in determine_encoding
gen 21 12:43:42 osmc python[291]:     self.update_raw()
gen 21 12:43:42 osmc python[291]:   File "/home/osmc/env/lib/python3.5/site-packages/yaml/reader.py", line 178, in update_raw
-- Logs begin at Sun 2019-01-20 16:17:01 CET, end at Mon 2019-01-21 12:45:30 CET. --
gen 20 16:17:02 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:40 osmc python[291]: INFO:gmusicapi.Mobileclient1:logged out
[... cut ...]
gen 21 12:43:43 osmc systemd[1]: Started Google Assistant-Voice.
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:version: 12.0.0
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:logging to: /home/osmc/.cache/gmusicapi/log/gmusicapi.log
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:initialized
gen 21 12:43:51 osmc python[1042]: INFO:gmusicapi.Mobileclient1:logged out

shivasiddharth, 2019-01-28:
If you are still stuck, try:
1. Open actions.py file
2.  Change line 88 from 
`with open(keywordfile,'r') as conf:`
to
`with open(keywordfile,'r' , encoding='utf8') as conf:`

Now try restarting again.