shivasiddharth/GassistPi #746: The volume gets low after playing audio from Youtube or Ganna.com

yash2197, 2019-01-16:
The assistant switches volume low to be heard but keeps the volume low as it is which makes it mute for conversation

shivasiddharth, 2019-01-16:
I am really unable to understand what your issue is.

yash2197, 2019-01-16:
Okay here's a brief....
The pi boots up and the assistant is initialised
It takes all the commands and the response is also in normal sound level
Now when a command to play music from Youtube or Ganna.com is issued voice is heard 'Getting Youtube link'
The song plays
Now the speaker volume is zero
If any command issued after this,the assistant processes it but no sound is heard in response
After manually starting the assistant I found out that the system volume of pi goes down to zero after wakeword is heard but it keeps it to zero after that

shivasiddharth, 2019-01-16:
Is there any reason for not using the issue template issue template ? Closing this issue for not using the issue template.

closed 2019-01-16 by shivasiddharth

yash2197, 2019-01-16:
Could you help me out here...
As I have to manually increase the volume of pi by logging in
Thanks

shivasiddharth, 2019-01-16:
Unless someone has played around with the scripts, what you are saying will not happen. Nowhere in the script i have set the assistant to control or change the master/pi volume. The volume dipping is confined only to the vlc player.

yash2197, 2019-01-16:
Could you specify where exactly that part is written in the code....
Would look at it and try to solve

yash2197, 2019-01-16:
The volume dipping part

shivasiddharth, 2019-01-16:
https://github.com/shivasiddharth/GassistPi/blob/d4f25e6fca13180070948a2783f03f3e4fb96c0f/src/main.py#L308