shivasiddharth/GassistPi #753: Youtube stream voice command always returns the same pop song no matter what I ask it to play from youtube

iamericha, 2019-01-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000a5436271

cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
```

### Describe the bug:
```
The GAssistPi recognizes when I tell it to stream a song from Youtube but no matter what song I ask it to play, it always plays the same pop song.  
[... 159 lines cut ...]
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Create your own action for button triple press
https://r4---sn-n4v7knlk.googlevideo.com/videoplayback?fvip=4&mm=31%2C26&gir=yes&c=WEB&signature=16B7D68036A2791D61156F5D058F4B1D25EDB10F.38A45011B8AE67A866C6ED8EC9ED68FD2069946D&id=o-AJh_J1lwuClI1NXZH3Hn49TnHMJeo-p2Tkbg8mfGz-Cg&dur=202.901&mt=1547863424&mv=m&ms=au%2Conr&source=youtube&clen=3292905&ip=24.4.156.199&keepalive=yes&mime=audio%2Fwebm&itag=251&ei=h4hCXPT7A8mAkga0xriQBg&requiressl=yes&mn=sn-n4v7knlk%2Csn-a5mlrn7k&key=yt6&initcwndbps=1991250&ipbits=0&pl=22&sparams=clen%2Cdur%2Cei%2Cgir%2Cid%2Cinitcwndbps%2Cip%2Cipbits%2Citag%2Ckeepalive%2Clmt%2Cmime%2Cmm%2Cmn%2Cms%2Cmv%2Cpl%2Crequiressl%2Csource%2Cexpire&txp=3511222&lmt=1541539157219733&expire=1547885799&ratebypass=yes
[02ed7040] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

ON_MUTED_CHANGED:
  {"is_muted": false}

```

### Attach the log file   
Log file can be found in /tmp/ directory  
*Log file not found in the directory unfortunately*

iamericha, 2019-01-19:
Hi aandroide,

You can see in the output I pasted above that it correctly heard me say “YouTube baby shark” though. That’s why I’m confused, I’ve tried a bunch of other things that used to work but I keep getting the same pop song each time.

aandroide, 2019-01-20:
The command correct is: play baby shark from youtube

iamericha, 2019-01-20:
Oh I see! Thanks!

aandroide, 2019-01-20:
3 days ago you changed youtube-dl open /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt and edit:
youtube_dl> = 23.11.2018
with
youtube_dl> = 17.01.2019
save and open the terminal and run:
pip install -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt
now it will work, but remember to say play 'song' from youtube

aandroide, 2019-01-21:
hello did you solve it?

iamericha, 2019-01-22:
Seems to work yes, thank you