shivasiddharth/GassistPi #754: TextToSpeechAPI problem

V1JCM, 2019-01-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000001e7d2e18

 cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Triend to use TextToSpeech API:
1) Ran update on 19.1.2019
2) Did like adviced:
_************************************************
### **USING GOOGLE CLOUD TEXT TO SPEECH**   
************************************************
**NOTE: GOOGLE CLOUD TEXT TO SPEECH HAS A LIMITED USAGE ACCESS. ONCE THE QUOTA IS EXHAUSTED, THE PROJECT WILL AUTOMATICALLY SWITCH TO gTTS**  
1. Go to the projects page on your Google Cloud Console-> https://console.cloud.google.com/project  
2. Select your project from the list.  
3. On the left top corner, click on the hamburger icon or three horizontal stacked lines.  
4. "From the API and services" option, select library and in the search bar type **text**, select "Cloud Text-to-Speech API" and click on "ENABLE".
5. In the API window, click on "Credentials" and then on "+ Create Credential".  
6. In the "Add credentials to your project" window, in step-1 under "Which API are you using?" drop down choose "Cloud Text-to-Speech API" and down below choose "No, I’m not using them". Then click on "What credentials do I need?"
7. In step-2 give your service acount a name and on the right in the "Role" drop down choose Project-->Owner and under "Key Type" select "JSON" and click "Continue".  
8. Copy the downloaded key and place it /home/pi/ directory **DO NOT RENAME**.   
9. Enter the path to the Key along with the key name Eg: /home/pi/xxxx.json  in the config.yaml file in the "Google_Cloud_TTS_Credentials_Path" field._  

3) GassistPI does not work any more, when ran manually I get:
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use M                                                                                                                usicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 44, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 21, in <module>
    logged_in = api.login(configuration['Gmusicapi']['email'],configuration['Gmusicapi']['password'] , configuration['Gmusicapi']['deviceid'])
KeyError: 'email'
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py

```

### Paste the contents of the terminal below
```
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use M                                                                                                                usicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 44, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 21, in <module>
    logged_in = api.login(configuration['Gmusicapi']['email'],configuration['Gmusicapi']['password'] , configuration['Gmusicapi']['deviceid'])
KeyError: 'email'
```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-01-19:
How did you update ?

V1JCM, 2019-01-22:
I had not update script, so I copied the scripts/update.sh from online and ran it. Seemed to do the trick?

V1JCM, 2019-01-23:
Ach, my bad. Should I reinstall?

shivasiddharth, 2019-01-24:
nah.. not required. Get the version file from master, change the feature number from 32 to something lower, save the file in the exact location as it is in the git. After that run the update.sh. I have designed the update script in such a way that you need not run the installer even for SDK updates.

V1JCM, 2019-01-26:
OK, thank you kindly, I got it working. However if I try to get text to speech working, by activating the API and adding this to config.yaml:
-------------------------------------
#Text to speech conversation
TextToSpeech:
  # Set the choice to:
  # 1. "gTTS" for quotaless text to speech conversion
  # 2. "Google Cloud" for Google Cloud text to sppech conversion. Free limit 4 Million Characters/month
  Choice: 'gTTS'
  Google_Cloud_TTS_Credentials_Path: "/home/pi/P<CEBSORED>1.json"
  #Voice for the "SAY" function or text to speech. Set it to 'Female' or 'Male' depending on your preference.
  #Note: This does not change Google's voice. this is only for the speech generated within custom actions.
  Voice_Gender: 'Male'
-------------------------------------
I get this error message:
-------------------------------------
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
<string>:2: GmusicapiWarning: login is deprecated and may break unexpectedly; prefer Mobileclient.oauth_login
INFO:gmusicapi.Mobileclient1:failed to authenticate
[031e44b8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[031efc80] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 44, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 193, in <module>
    if configuration['Voice_Custom_Actions']=='Male':
KeyError: 'Voice_Custom_Actions'
-------------------------------------

shivasiddharth, 2019-01-26:
Your actions.py is outdated.

V1JCM, 2019-01-26:
Indeed! All my GassistPI files are outdated, 15.12.2018, except for update.sh.

Something seems to go wrong when I run update.sh. I get (same thing without "sudo"):
---------------------------
 sudo /home/${USER}/GassistPi/scripts/update.sh               

This script must run as pi, trying to change user...
Löytyi:1 http://archive.raspberrypi.org/debian stretch InRelease
Löytyi:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Luetaan pakettiluetteloita... Valmis
Luetaan pakettiluetteloita... Valmis
Muodostetaan riippuvuussuhteiden puu
Luetaan tilatiedot... Valmis
jq is already the newest version (1.5+dfsg-1.3).
0 päivitetty, 0 uutta asennusta, 0 poistettavaa ja 51 päivittämätöntä.
Checking for updates................

URL transformed to HTTPS due to an HSTS policy
--2019-01-26 15:42:02--  https://raw.githubusercontent.com/shivasiddharth/GassistPi/master/version
Selvitetään osoitetta raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.244.133
Yhdistetään palvelimeen raw.githubusercontent.com (raw.githubusercontent.com)|151.101.244.133|:443... yhdistetty.
HTTP-pyyntö lähetetty, odotetaan vastausta... 200 OK
Pituus: 76 [text/plain]
Tallennetaan kohteeseen ”/tmp/version”

/tmp/version        100%[===================>]      76  --.-KB/s    in 0s

2019-01-26 15:42:03 (23,6 MB/s) - ”/tmp/version” tallennettu [76/76]

2
The project is already uptodate........
----------------------------------------------

/home/${USER}/GassistPi/version is like this:
{
  "Version":{
    "SDK":051,
    "Features":017,
    "Revision":002
  }
}

shivasiddharth, 2019-01-26:
fixed the errata in the update script.

V1JCM, 2019-01-27:
Success! Cheers, mate!

However, custom phrases are still female "no I am forever alone"), even though I have set the voice to "male". Any ideas?

shivasiddharth, 2019-01-27:
Some languages have only female voice. What language are you usin ?

shivasiddharth, 2019-01-27:
Can you check the case of the letters that you have used? it should be "Male" with "M" in caps.

V1JCM, 2019-01-27:
I got it like this:
  Voice_Gender: 'Male'

But something else went wrong as I rebooted from the console, nothing seems to work. GassistPI does not even start. I type :
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py

and I get this:
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 45, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 9, in <module>
    from youtube_search_engine import youtube_search
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 13, in <module>
    configuration = yaml.load(conf)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/constructor.py", line 35, in get_single_data
    node = self.get_single_node()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/home/pi/env/lib/python3.5/site-packages/yaml/composer.py", line 110, in compose_sequence_node
    while not self.check_event(SequenceEndEvent):
  File "/home/pi/env/lib/python3.5/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/home/pi/env/lib/python3.5/site-packages/yaml/parser.py", line 393, in parse_block_sequence_entry
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block collection
  in "/home/pi/GassistPi/src/config.yaml", line 22, column 7
expected <block end>, but found '<scalar>'
  in "/home/pi/GassistPi/src/config.yaml", line 23, column 24

Then I ran:
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/Extras/gmusicauth.py
But to no avail.

shivasiddharth, 2019-01-27:
there is something wrong with your config.yaml entries in line 22 and 23.

V1JCM, 2019-01-27:
Haa! Now I realise, my bad! See what I added "Antti's" is grammatically correct but otherwise totally false, as there is ' -symbol!:

  question:
    1:
      - 'Who are you'
    2:
      - 'Do you have a family'
    3:
      - 'Who lives in this room'
  answer:
    1:
      - 'I am not the one you expect to be'
      - 'Who do you think I am'
      - 'What bothers you'
    2:
      - 'Nope i am forever alone'
      - 'The whole world is my family'
    3:
      - 'Antti lives here'
      - 'This is Antti's room'

Thank you once again for help, I appreciate it. I fixed the error and rebooted, all seems to be fine. However, now Google text so speech API is not working. Anttis room is made by the machine voice you created earlier.

shivasiddharth, 2019-01-27:
Have you added the credentials in the config.yaml file ? Also check the Choice for text to speech that you have entered.

shivasiddharth, 2019-01-28:
> Set the choice to:
> 1. "gTTS" for quotaless text to speech conversion
> 2. "Google Cloud" for Google Cloud text to sppech conversion. Free limit 4 Million Characters/month
> Choice: 'gTTS'

You have set the choice to gTTS, so obviously gTTS will only work.

shivasiddharth, 2019-01-29:
No worries..

closed 2019-01-29 by shivasiddharth