shivasiddharth/GassistPi #762: News and podcast issue

DewGew, 2019-01-23:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial          : 000000007c45309a

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
  
```

### Describe the bug:
```
[... 172 lines cut ...]
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
see above

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/2786046/GassistPi.log)

DewGew, 2019-01-23:
I read somewhere that libmpg123.so is needed for listen to news/podcasts. So I installed that with:
`sudo apt install libmpg123-dev`. Now GA is accepting my request but can't hear any news. I get this in terminal:
```
ON_MUTED_CHANGED:
  {"is_muted": true}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

Say "OK, Google" then speak, or press Ctrl+C to quit...

Connected with result code 0
Message from MQTT: listen to the news
ON_CONVERSATION_TURN_STARTED

E0123 07:01:35.056994892    3296 fork_posix.cc:63]           Fork support is only compatible with the epoll1 and poll polling strategies
ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MEDIA_TRACK_LOAD:
{
  "metadata": {
    "album": "Nyheter från Ekot",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "title": "Ekot"
  },
  "track_type": 1
}

ON_MEDIA_TRACK_PLAY:
{
  "metadata": {
    "album": "Nyheter från Ekot",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "duration_ms": 536870912,
    "title": "Ekot"
  },
  "position_ms": 0,
  "track_type": 1
}

ON_MEDIA_TRACK_LOAD:
{
  "metadata": {
    "album": "Nyheter från Ekot",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "title": "Ekot"
  },
  "track_type": 2
}

ON_MEDIA_STATE_ERROR

ON_MEDIA_TRACK_LOAD:
  {"metadata": {"title": "closingTts"}, "track_type": 1}

ON_MEDIA_TRACK_PLAY:
{
  "metadata": {
    "duration_ms": 536870912,
    "title": "closingTts"
  },
  "position_ms": 0,
  "track_type": 1
}

ON_MEDIA_STATE_IDLE

```

DewGew, 2019-01-23:
#763

shivasiddharth, 2019-01-23:
Do you have a source where it says that you need libmpg123-dev ? i am unable to locate a source. Also i see the ON_MEDIA_STATE_ERROR which means that the assistant is unable to play the news.

DewGew, 2019-01-23:
I found it here: https://github.com/googlesamples/assistant-sdk-python/issues/260#issuecomment-399541722

shivasiddharth, 2019-01-23:
Go into actions console and select the registered model.  Click on supported traits and unselect OnOff and StartStop.  Restart your assistant and try news.

DewGew, 2019-01-24:
I tested to disable OnOff and StartStop traits in actions console. But I got same error:
```
device_model_id: gassistpi-e8556-gassistpi-ggnze6
device_id: 182567F77254FB7E7775870EAFD387DA

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

Say "OK, Google" then speak, or press Ctrl+C to quit...

ON_MEDIA_STATE_IDLE

Connected with result code 0
Message from MQTT: listen to the news
ON_CONVERSATION_TURN_STARTED

E0124 07:35:55.485407181     964 fork_posix.cc:63]           Fork support is only compatible with the epoll1 and poll polling strategies
ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MEDIA_TRACK_LOAD:
{
  "metadata": {
    "album": "Morgonekot: nyheter och reportage",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "title": "Ekot"
  },
  "track_type": 1
}

ON_MEDIA_TRACK_PLAY:
{
  "metadata": {
    "album": "Morgonekot: nyheter och reportage",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "duration_ms": 536870912,
    "title": "Ekot"
  },
  "position_ms": 0,
  "track_type": 1
}

ON_MEDIA_TRACK_LOAD:
{
  "metadata": {
    "album": "Morgonekot: nyheter och reportage",
    "album_art": "http://lh5.googleusercontent.com/proxy/GEel6tE36_EtwoxmgAxyeiTp_Af9T94D6msxY46krwxHK3fq15JnoH-pJWjylgaUIvmUpCmn5Ubgm2O1mzB1wl1_4Ajat7Bzff8n1-Y_u2SHR86rH80Kk31lTuPG81yon2fMkr1xpjI2S93SR1idLkZl4Ifj_Q=s128",
    "title": "Ekot"
  },
  "track_type": 2
}

ON_MEDIA_STATE_ERROR

ON_MEDIA_TRACK_LOAD:
  {"metadata": {"title": "closingTts"}, "track_type": 1}

ON_MEDIA_TRACK_PLAY:
{
  "metadata": {
    "duration_ms": 536870912,
    "title": "closingTts"
  },
  "position_ms": 0,
  "track_type": 1
}

ON_MEDIA_STATE_IDLE
```

closed 2019-01-24 by DewGew