shivasiddharth/GassistPi #791: Hifiberry + USB Mic

AlexHowser, 2019-01-29:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Raspberry PI zero w`
cat /proc/cpuinfo  
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000f975c532

cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
I am trying to set up my assistant using an usb mic and a hifiberry-dac, recording and playback over mplayer and arecord work fine. I did not use any audio config script since i did not know wich one to use.. When i try to finish the setup i get: 

Testing the installed google assistant. Make a note of the generated Device-Id
INFO:root:Connecting to embeddedassistant.googleapis.com
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 10, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 351, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1264, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 780, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2572, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Invalid sample rate [PaErrorCode -9997]

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Testing the installed google assistant. Make a note of the generated Device-Id
INFO:root:Connecting to embeddedassistant.googleapis.com
Traceback (most recent call last):
  File "/home/pi/env/bin/googlesamples-assistant-pushtotalk", line 10, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/pushtotalk.py", line 351, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1264, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 780, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2572, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Invalid sample rate [PaErrorCode -9997]

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-01:
Use the USB-DAC option.

AlexHowser, 2019-03-15:
Got it working, Hardware ID was 0,0
Thanks!

shivasiddharth, 2019-03-19:
Closing this issue.

closed 2019-03-19 by shivasiddharth