shivasiddharth/GassistPi #800: Youtube API Key not recognized

JoshSullivan11, 2019-02-03:
### Board and OS details:    
pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000ea0a2c32
pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:

"no longer available on youtube" after making any request. I followed steps given in the Youtube API setup and Generating API key, but nothing is working. I am completely new to the Raspberry PI so please bear with me.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.
 
Mine does not output to the terminal that I can see, should it? If so how would i get it to do this?

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/2825058/GassistPi.log)

shivasiddharth, 2019-02-03:
Run the following in the terminal and post the output.
```
source env/bin/activate
pip freeze youtube_dl
```

JoshSullivan11, 2019-02-03:
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip freeze youtube_dl
aftership==0.2
appdirs==1.4.3
asn1crypto==0.24.0
beautifulsoup4==4.6.3
bs4==0.0.1
cachetools==3.0.0
casttube==0.2.0
certifi==2018.11.29
cffi==1.11.5
chardet==3.0.4
click==6.7
cryptography==2.4.2
decorator==4.3.0
enum34==1.1.6
feedparser==5.2.1
future==0.17.1
futures==3.1.1
gmusicapi==11.1.1
google-api-core==1.7.0
google-api-python-client==1.7.7
google-assistant-grpc==0.2.1
google-assistant-library==1.0.1
google-assistant-sdk==0.5.1
google-auth==1.6.2
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.2.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.5.5
googletrans==2.3.0
gpiozero==1.4.1
gpsoauth==0.4.1
grpcio==1.17.1
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.12.0
idna==2.8
ifaddr==0.1.6
ipaddress==1.0.22
kodi-json==1.0.0
langdetect==1.0.7
lxml==4.2.5
MechanicalSoup==0.11.0
mock==2.0.0
mutagen==1.42.0
numpy==1.15.4
oauth2client==4.1.3
oauthlib==2.1.0
pafy==0.5.4
paho-mqtt==1.4.0
pathlib2==2.3.3
pbr==5.1.1
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.6.1
psutil==5.4.8
pushbullet.py==0.11.0
pyasn1==0.4.4
pyasn1-modules==0.2.2
PyAudio==0.2.11
PyChromecast==2.3.0
pycparser==2.19
pycryptodomex==3.7.2
pyOpenSSL==18.0.0
python-dateutil==2.7.5
python-magic==0.4.15
python-vlc==3.0.4106
pytz==2018.9
PyYAML==3.13
requests==2.21.0
requests-oauthlib==1.0.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.12.0
sounddevice==0.3.12
spidev==3.2
spotipy==2.4.4
tenacity==4.12.0
uritemplate==3.0.0
urllib3==1.24.1
validictory==1.1.2
websocket-client==0.54.0
youtube-dl==2018.11.23
zeroconf==0.21.3
(env)

shivasiddharth, 2019-02-03:
I edited the comment please check the updated comment.

JoshSullivan11, 2019-02-03:
This just made everything stop working... google assistant won't start now. I tried starting and stopping manually but nothing works. I had it set to start on boot automatically originally. I tried updating it and it is up to date. Don't know what else to do unless i try a fresh install, unless you have any suggestions

shivasiddharth, 2019-02-03:
There is no way that this will crash the assistant. manually start the assistant and post the terminal output.

divijbajaj5, 2019-02-03:
If I have set my assistant to headless autostart at boot, How can I remove this service to manually start my assistant?

shivasiddharth, 2019-02-03:
@divijbajaj5 Do not Hijack threads, this is against Git etiquette. Use the Gitter for non-issue posts.

divijbajaj5, 2019-02-03:
> @divijbajaj5 Do not Hijack threads, this is against Git etiquette. Use the Gitter for non-issue posts.

Sorry, I have deleted the comment

JoshSullivan11, 2019-02-04:
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'gassitpi-j-gassitpi-j-n6aqkd'
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-04 17:45:03,690 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 49, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 237, in <module>
    if configuration['Pushbullet']['Pushbullet_API_KEY']!='ENTER YOUR PUSHBULLET KEY HERE':
KeyError: 'Pushbullet'

JoshSullivan11, 2019-02-05:
thanks, that was the issue! much appreciated

nicusorflorin, 2019-02-06:
@shivasiddharth i got a similar issue, already compared my actions.py and config.yaml with the repository master code, and it the same, this is what i get from "pip freeze youtube_dl"
```
automationhat==0.1.0
beautifulsoup4==4.5.3
blinker==1.3
blinkt==0.1.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
drumhat==0.1.0
enum34==1.1.6
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gpiozero==1.4.1
html5lib==0.999999999
idna==2.2
inflect==2.1.0
ipaddress==1.0.17
itsdangerous==0.24
jaraco.itertools==4.4.1
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
lxml==3.7.1
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
more-itertools==5.0.0
mote==0.0.4
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
pika==0.13.0
Pillow==4.0.0
pyasn1==0.1.9
PyAudio==0.2.11
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
samsungctl==0.7.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
touchphat==0.0.1
twython==3.4.0
urllib3==1.19.1
webencodings==0.5
Werkzeug==0.11.15
wol==0.2
wolframalpha==3.0.1
xmltodict==0.11.0
youtube-dl==2019.1.30.1
```
Got any clues for me? :(

shivasiddharth, 2019-02-06:
Run:

```
source env/bin/activate
pip install youtube_dl==2019.1.30
```
youtube-dl==2019.1.30.1 was released only hours back, could be broken.

nicusorflorin, 2019-02-07:
still same issue, it didn't seem to help, restarted the assistant and all :( (not sure if related but i also have issue with google music it doesn't work and the log only says "stopping vlc", i tracked the function down and put breakpoints in all places that it is called and debugged it, but none are triggered, which is mightly strange, i will post a separate bug for this after fixing the youtube issue if it still happens)

aandroide, 2019-02-07:
has been released a new update, run:
source env / bin / activate
pip install youtube_dl == 2019.02.08

remember to enter first in: "source env / bin / activate"

nicusorflorin, 2019-02-08:
@shivasiddharth the same as @JoshSullivan11 : "no longer available on youtube" after making any request. 
PS: i tried the new update, same issue

aandroide, 2019-02-08:
did you enter the youtube credentials in the config.yaml file? and have you made sure your bee key is active?

shivasiddharth, 2019-02-08:
@nicusorflorin  Pull the latest actions.py file from master. I have set it to print the exception. Now we can see as to whats going wrong.

nicusorflorin, 2019-02-08:
Ok, problem fixed, it seems google deactivated the youtube Api for my project for inactivity. sorry for the bother. will double check my google play music , before submitting that issue

shivasiddharth, 2019-02-08:
@nicusorflorin Please dont delete the expired key. Use it in the project and also use the updated actions.py script. We can take a look at the error and catch that. Can you please change the key after reporting the error thats being printed.

nicusorflorin, 2019-02-08:
ok. will do

aandroide, 2019-02-08:
when you look for a song how do you pronounce the command?
it should be: hey google, "play song from youtube

nicusorflorin, 2019-02-08:
here you go:
```

ON_MUTED_CHANGED:
  {"is_muted": false}

Waiting for commands
ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play The Islander by Nightwish from YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
[?25l 
Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
[?25h the islander by nightwish
<HttpError 403 when requesting https://www.googleapis.com/youtube/v3/search?key=AIzaSyB2LIMCjC9TO-Z6slD7kiHUoqwB9UOeKOk&q=the+islander+by+nightwish&alt=json&part=id%2Csnippet returned "Access Not Configured. YouTube Data API has not been used in project 893332908849 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=893332908849 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.">
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
[?25l 
Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
[?25h ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

shivasiddharth, 2019-02-08:
Closing this issue then. @nicusorflorin  You can open up a separate issue for gmusic just in case.

shivasiddharth, 2019-02-08:
@aandroide , Can you use an old version of youtube_dl and share the error that is getting printed? Thanks in advance.

closed 2019-02-08 by shivasiddharth

aandroide, 2019-02-09:
@shivasiddharth  I tried to uninstall the new version and install the old version, but oddly it works and I do not get errors.

shivasiddharth, 2019-02-09:
can you some version from September or October release ?

aandroide, 2019-02-09:
youtube-dl 2018.06.14 this version work!!!

shivasiddharth, 2019-02-09:
oh my god... never mind then. thanks