shivasiddharth/GassistPi #808: the assistant does not start with the new update

aandroide, 2019-02-04:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
(env) pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000a91ab7c2
(env) pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
(env) pi@raspberrypi:~ $ 

### Describe the bug:
```

the assistant does not start with the new update
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 45, in <module>
    from google.cloud import speech
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech.py", line 19, in <module>
    from google.cloud.speech_v1 import SpeechClient
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/__init__.py", line 17, in <module>
    from google.cloud.speech_v1.gapic import speech_client
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/gapic/speech_client.py", line 33, in <module>
    from google.cloud.speech_v1.gapic.transports import speech_grpc_transport
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/gapic/transports/speech_grpc_transport.py", line 20, in <module>
    from google.cloud.speech_v1.proto import cloud_speech_pb2_grpc
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2_grpc.py", line 4, in <module>
    from google.cloud.speech_v1.proto import (
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py", line 46, in <module>
    google_dot_rpc_dot_status__pb2.DESCRIPTOR,
TypeError: __init__() got an unexpected keyword argument 'serialized_options'

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'replace this with the model id'

```

### Paste the contents of the terminal below
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 45, in <module>
    from google.cloud import speech
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech.py", line 19, in <module>
    from google.cloud.speech_v1 import SpeechClient
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/__init__.py", line 17, in <module>
    from google.cloud.speech_v1.gapic import speech_client
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/gapic/speech_client.py", line 33, in <module>
    from google.cloud.speech_v1.gapic.transports import speech_grpc_transport
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/gapic/transports/speech_grpc_transport.py", line 20, in <module>
    from google.cloud.speech_v1.proto import cloud_speech_pb2_grpc
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2_grpc.py", line 4, in <module>
    from google.cloud.speech_v1.proto import (
  File "/home/pi/env/lib/python3.5/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py", line 46, in <module>
    google_dot_rpc_dot_status__pb2.DESCRIPTOR,
TypeError: __init__() got an unexpected keyword argument 'serialized_options'

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-04:
I am unable to replicate this issue. I leave this open to see if someone else reports this. As far as I know people have got the latest update running properly. Seems like a localized issue. Run pip freeze within environment and share the output.

shivasiddharth, 2019-02-11:
Issue resolved...

closed 2019-02-11 by shivasiddharth

ganeshArwat, 2022-02-18:
i have also same issue

ganeshArwat, 2022-02-20:
how can i slove it