shivasiddharth/GassistPi #811: interpreter fail

jcksnmdy, 2019-02-06:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000061002412
---------------------------------------------------------
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
### Describe the bug:
```
When asking for any interpreter, i get a segmentation fault 
[... 164 lines cut ...]
2019-02-05 22:24:44,879 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-02-05 22:24:44,904 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-02-05 22:24:50,811 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-02-05 22:25:25,077 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2019-02-05 22:25:25,097 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2019-02-05 22:25:28,975 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2019-02-05 22:45:40,990 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-02-05 22:45:41,013 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-02-05 22:45:46,675 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-02-05 22:46:31,829 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2019-02-05 22:46:31,851 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2019-02-05 22:46:33,821 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2019-02-05 22:47:07,509 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2019-02-05 22:47:07,529 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2019-02-05 22:47:11,817 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2019-02-05 22:48:59,593 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-02-05 22:48:59,617 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-02-05 22:49:04,828 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-02-05 22:50:01,319 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2019-02-05 22:50:01,342 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2019-02-05 22:50:06,464 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
2019-02-05 22:50:41,921 : DEBUG : Making request: POST https://oauth2.googleapis.com/token
2019-02-05 22:50:41,941 : DEBUG : Starting new HTTPS connection (1): oauth2.googleapis.com:443
2019-02-05 22:50:45,501 : DEBUG : https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
```

shivasiddharth, 2019-02-06:
Did you go through this section https://github.com/shivasiddharth/GassistPi#using-live-translator-or-interpreter and follow the steps ?

jcksnmdy, 2019-02-06:
Yes

shivasiddharth, 2019-02-06:
Replace your existing main.py with the updated file from master. Give the same command and paste the error output. Also i would like to know as to how many times you are getting the segmentation error. Like you are getting it at all times when you are giving the interpreter command or sometimes only

jcksnmdy, 2019-02-07:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "start Spanish interpreter"}

High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: gcloud.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of gcloud.mp3 finished.
E0206 21:36:50.289076995    1308 fork_posix.cc:63]           Fork support is only compatible with the epoll1 and poll polling strategies
Fatal Python error: Segmentation fault

Thread 0x655ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x62dff470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76fcd010 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1006 in main
  File "/home/pi/GassistPi/src/main.py", line 1014 in <module>
Segmentation fault
-------------------------------------------------------------------
It happens all the time

shivasiddharth, 2019-02-07:
What is the audio that you are getting ?

> High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
> 	version 1.23.8; written and copyright by Michael Hipp and others
> 	free software (LGPL) without any warranty but with best [...]

What does it say ?

jcksnmdy, 2019-02-08:
What do you mean

shivasiddharth, 2019-02-08:
> High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
> version 1.23.8; written and copyright by Michael Hipp and others
> free software (LGPL) without any warranty but with best wishes [...]

It is an audio of text to speech conversion. what do you hear? or What does it say? or What audio does the text to speech synthesizer play ?

jcksnmdy, 2019-02-08:
After i ask for the interpreter i say "hi, how are you?" Then it says segmentation error without translating or showing what i said.

shivasiddharth, 2019-02-08:
Attach the main.py what you have here.

jcksnmdy, 2019-02-10:
```
#!/usr/bin/env python

# Copyright (C) 2017 Google Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from __future__ import print_function
import faulthandler
faulthandler.enable()
from kodijson import Kodi, PLAYER_VIDEO
try:
    import RPi.GPIO as GPIO
except Exception as e:
    if str(e) == 'No module named \'RPi\'':
        GPIO = None
import argparse
import json
import os.path
import pathlib2 as pathlib
import os
import subprocess
import re
import psutil
import logging
import time
import random
import snowboydecoder
import sys
import signal
import requests
import io
import google.oauth2.credentials
from google.assistant.library import Assistant
from google.assistant.library.event import EventType
from google.assistant.library.file_helpers import existing_file
from google.assistant.library.device_helpers import register_device
from google.cloud import speech
from google.cloud.speech import enums
from google.cloud.speech import types
import paho.mqtt.client as mqtt
from actions import say
from actions import trans
from actions import Action
from actions import YouTube_No_Autoplay
from actions import YouTube_Autoplay
from actions import stop
from actions import radio
from actions import ESP
from actions import track
from actions import feed
from actions import kodiactions
from actions import mutevolstatus
from actions import gmusicselect
from actions import refreshlists
from actions import chromecast_play_video
from actions import chromecast_control
from actions import kickstarter_tracker
from actions import getrecipe
from actions import hue_control
[... 921 lines cut ...]
                                    device_model_id, device_id, args.nickname)
                    pathlib.Path(os.path.dirname(args.device_config)).mkdir(
                        exist_ok=True)
                    with open(args.device_config, 'w') as f:
                        json.dump({
                            'last_device_id': device_id,
                            'model_id': device_model_id,
                        }, f)
                else:
                    print(WARNING_NOT_REGISTERED)

            for event in events:
                if event.type == EventType.ON_START_FINISHED and args.query:
                    assistant.send_text_query(args.query)
                self.process_event(event)

        if custom_wakeword:
            self.detector.terminate()

if __name__ == '__main__':
    try:
        Myassistant().main()
    except Exception as error:
        logger.exception(error)
```

shivasiddharth, 2019-02-10:
pull the latest main.py from master. whatever be the error that will print it. manually start the assistant and share the error.

jcksnmdy, 2019-02-14:
It says I don't have a billing method so i will work on that.

shivasiddharth, 2019-02-14:
https://github.com/shivasiddharth/GassistPi/issues/811#issuecomment-460971278  That was the first thing i asked, if you followed the steps. You clearly have not, but said that you did. Closing this issue.

closed 2019-02-14 by shivasiddharth

jcksnmdy, 2019-02-15:
It turned out that i had a billing method and it ran out when i was tesing, sorry. Now i am getting this error
```
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-14 23:27:17,482 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
E0214 23:27:20.690786757    1172 fork_posix.cc:63]           Fork support is only compatible with the epoll1 and poll polling strategies
[01f54f40] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
E0214 23:27:20.778306324    1172 fork_posix.cc:63]           Fork support is only compatible with the epoll1 and poll polling strategies
[01f01060] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1024, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9996] Invalid input device (no default output device)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1026, in <module>
    logger.exception(error)
NameError: name 'logger' is not defined
```