shivasiddharth/GassistPi #813: google assistant goes into listening mode automatically

neha-s-deshpande, 2019-02-06:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
pi@raspberrypi:~/GassistPi/src $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000003212288c
pi@raspberrypi:~/GassistPi/src $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs".   
```

### Describe the bug:
```
I am running GassistPi at boot up on Raspberry Pi 3. I have added custom wake words to it with custom_wakeword flag to enable and it is running fine. The problem is that it gets into listening mode automatically even when no hotword is detected.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-06:
These are false positive triggers. How did you record the models? You can adjust the sensitivity in the main.py https://github.com/shivasiddharth/GassistPi/blob/16527ec1408977dc0f83028f1a809c159ff82489/src/main.py#L188

neha-s-deshpande, 2019-02-06:
Thanks. That worked..
But when I autostart Gassistpi at bootup, it takes too much time to start.

neha-s-deshpande, 2019-02-06:
I recorded the models with snowboy with sensitivity of 0.5

shivasiddharth, 2019-02-06:
Cant do much about the boot time, there are too many factors that we have no control of. Did you record the models using the pi itself with the audio setup that you are using for the assistant or did you use some other PC?

neha-s-deshpande, 2019-02-06:
I recorded the models on Raspberry Pi. I used the public model on snowboy, recorded it again with my voice and then added in script

neha-s-deshpande, 2019-02-06:
Hi, I am trying to add custom commands to google assistant using webhooks. But it requires an agent to invoke which is created using dialogflow. Is there any other way to implement custom commands without need to invoke agent as the responses to queries are dynamic?

shivasiddharth, 2019-02-06:
If you are using the Pi itself then checkout the custom-actions https://developers.google.com/assistant/sdk/guides/library/python/extend/custom-actions

shivasiddharth, 2019-02-11:
Issue Resolved, closing this thread.

closed 2019-02-11 by shivasiddharth