shivasiddharth/GassistPi #818: Google Music Not playback doesn't work

nicusorflorin, 2019-02-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```

cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 0000000032cbad72
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
When asked to play something, it fails to do so, ran the authenticator and all (i uninstalled pulseaudio, not sure if that might be related)
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs by artist manowar from Google Music"}

stopping vlc
manowar
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
[?25l 
Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
[?25h Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
................................ (a couple hundreds of these :) )
Artist not found
bash: line 1:  2992 Segmentation fault      env "PYTHONUNBUFFERED"="1" "PYTHONPATH"="/home/pi/GassistPi/src" "PYCHARM_HOSTED"="1" "JETBRAINS_REMOTE_RUN"="1" "PYTHONIOENCODING"="UTF-8" /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'crazypiassistant2' --device_model_id 'crazypiassistant2-crazyassistant-opzeww'

Process finished with exit code 139
```

### Attach the log file   
Log file can be found in /tmp/ directory

rockman55, 2019-02-08:
I have a similar problem, when I tell Google to Play either a Playlist or all songs, it makes a reference back that it is doing that, no songs are played and the assistant restarts. I ran the assistant manually and am posting the terminal output in the attached file.

[manual_start.txt](https://github.com/shivasiddharth/GassistPi/files/2846970/manual_start.txt)

shivasiddharth, 2019-02-08:
@rockman55 , you have incorrectly manually started the assistant. Infact your assistant did not even start.

shivasiddharth, 2019-02-08:
@nicusorflorin , What happens if you ask it to play all the songs?

rockman55, 2019-02-08:
To Nick, when I ask Google to "Play all the songs form Google Music", the response is "looking for your songs", then a pause and the assistant restarts with the ding and 'Hello'.

rockman55, 2019-02-08:
to shivasiddaharth, I followed the instructions on manually restarting the assistant from the Gitter Lobby and then copied the output from the terminal. I can do it again, but please give me the steps to properly execute that. Thanks

shivasiddharth, 2019-02-09:
There is no step really. Just follow this: https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant and enter your model id in the mentioned place within quotes.

rockman55, 2019-02-09:
This is the command I am using to Manuall Start the assistant: 
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --customgoogle
I have tried another Model ID (not shown here) with the same error codes. Where exactly do I find the model id? I got this number from the Credentials page for my project, customgoogle. 
Thanks

shivasiddharth, 2019-02-09:
`/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'replace this with the model id'` This is the command to start the assistant manually. Only part that you have to change is `replace this with the model id` model id is the alphanumeric characters that you got from actions console. If you open /home/pi/ directory, there will be a text file where also you can find the model id. Here is a snapshot of where your model-id will be 

[image]

rockman55, 2019-02-09:
Ok, I was using the wrong format to manually start the assistant. Here is a fresh copy of the terminal screen after manual start. Hope this helps you, help me to determine what is wrong with Google Music.  Thanks for your great project and help.
[manual_start_2.txt](https://github.com/shivasiddharth/GassistPi/files/2848105/manual_start_2.txt)

rockman55, 2019-02-09:
Some additional info, I tried to play first playlist from Google Music and here is what appeared in the terminal:
ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs from the first playlist in Google Music"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of female-say.mp3 finished.
Segmentation fault
pi@customgoogle:~ $

aandroide, 2019-02-09:
The command  correct is: play song from google music

shivasiddharth, 2019-02-09:
The correct syntax should contain from Google Music and not in Google Music. I have corrected the README now. Try: "Hey Google, Play All the songs from google music".

rockman55, 2019-02-09:
Tried that command and here is what I see in terminal:
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all the songs from Google Music"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
Segmentation fault
pi@customgoogle:~ $

rockman55, 2019-02-09:
To aandroide I tried your command and here is the outcome:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play song from Google Music"}

stopping vlc
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I'm a mess from Google Music"}

stopping vlc
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
----------
I was trying to play a song by Ed Sheeran, called I'm A Mess.
Thanks

shivasiddharth, 2019-02-10:
pull the latest main.py from master. whatever be the error that will print it. manually start the assistant and share the error.

rockman55, 2019-02-11:
The file manual_start.txt is the terminal after doing a manual start.
[manual_start.txt](https://github.com/shivasiddharth/GassistPi/files/2849804/manual_start.txt)

shivasiddharth, 2019-02-11:
@rockman55 This file does not help. You need to give the Google music command only then, it will print the error and then i can take a look.

Phoenix260, 2019-02-11:
I think I'm getting the same error. May as well share mine too if it helps. I've updated to your latest main.py and here is the error:

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all the songs from Google Music"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Segmentation fault
GetStreamUrl: 403 Client Error: Forbidden for url: https://mclients.googleapis.com/music/mplay?opt=hi&net=mob&pt=e&slt=1549904139714&sig=zgg1OkvcyvI1y4zIsO_9BmPFVk4&songid=d7a172fd-fadb-31b9-b3af-e247f7c541ef&hl=en_US&dv=0&tier=fr
(requests kwargs: {'method': 'GET', 'url': 'https://mclients.googleapis.com/music/mplay', 'allow_redirects': False, 'headers': {'X-Device-ID': 'B827EBBEEC1C', 'Authorization': '<omitted>'}, 'params': {'opt': 'hi', 'net': 'mob', 'pt': 'e', 'slt': '1549904139714', 'sig': b'zgg1OkvcyvI1y4zIsO_9BmPFVk4', 'songid': 'd7a172fd-fadb-31b9-b3af-e247f7c541ef', 'hl': 'en_US', 'dv': 0, 'tier': 'fr'}})
(response was: '<HTML>\n<HEAD>\n<TITLE>Forbidden</TITLE>\n</HEAD>\n<BODY BGCOLOR="#FFFFFF" TEXT="#000000">\n<H1>Forbidden</H1>\n<H2>Error 403</H2>\n</BODY>\n</HTML>\n')
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes
[src/libmpg123/readers.c:1182] error: Cannot open file /tmp/female-say.mp3: No such file or directory
main: [src/mpg123.c:732] error: Cannot open /tmp/female-say.mp3: File access error. (code 22)
Fatal Python error: Segmentation fault

Thread 0x648ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 474 in start_detector
  File "/usr/local/lib/python3.7/threading.py", line 865 in run
  File "/usr/local/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/local/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x66aff470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 247 in pushbutton
  File "/usr/local/lib/python3.7/threading.py", line 865 in run
  File "/usr/local/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/local/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76fcd510 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1015 in main
  File "/home/pi/GassistPi/src/main.py", line 1023 in <module>
Segmentation fault

```

shivasiddharth, 2019-02-11:
Replace your existing mediaplayer.py file with the updated file from master. If the issue has not been fixed, please feel free to re-open.

closed 2019-02-11 by shivasiddharth

rockman55, 2019-02-12:
Thanks for the fix, I can now select an artist and it plays. However how do you stop playing or move to another song? I tried to say Stop, or Hey Google Stop, Hey Google Pause, and nothing works. Below is  the bottom of the terminal I was in after starting to play this song. I cannot get back to the top of the terminal since there is an enormous amount of "Artist Not Found":
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
[031d91e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

[6f1a6190] prefetch stream error: unimplemented query (264) in control
[626355d0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[68f91bc0] prefetch stream error: unimplemented query (264) in control

Thanks again for all your help.

shivasiddharth, 2019-02-12:
Comment out the print command for "Artist not found". If you followed the README syntax, then switching tracks, pause, play, stop everything should work.

rockman55, 2019-02-12:
I could not get any songs to pause or play next track. I follow the instructions in the read me file and I have updated the two files main.py and mediaplayer.py. When I went to the actions.py file to comment out "Artist not found", I put a # in front of the  print("Album not found"). This causes a syntax error, so not sure how to comment it out. Do you need to see the terminal output when I execute the Hey Google, Pause Music command?

commit d66cf843 (2019-02-11): Update mediaplayer.py

Fixes #818 . Fixes Google Music Issue.