shivasiddharth/GassistPi #820: Hotword recognition not work

chrisf95, 2019-02-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 0000000017b95439

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"   
```

### Describe the bug:
```
I installed GassistPi on a Pi Zero W using a usb mic and a usb speaker (they work fine using aplay and arecord command)
I activated the autoboot, but i still continue to activate automatically and say "hi" and I can't do nothing
the hotword recognition is enabled, but don't work (neither OK google or custom ones)
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-08 19:10:18,049 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[018f0608] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[018fc198] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Initializing GPIOs 5 and 6 for assistant activity indication
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 960, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 814, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 190, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1264, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 780, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2572, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-09:
Post the output of following:

```
sudo nano /etc/asound.conf
sudo nano /.asoundrc
aplay -l
arecord -l
```

chrisf95, 2019-02-09:
I don't have a asound.conf file in etc

/home/pi/.asound
pcm.!default {
    type asym
    capture.pcm "mic"
    playback.pcm "speaker"
}
pcm.mic {
    type plug
    slave {
        pcm "hw:1,0"
    }
}
pcm.speaker {
    type plug
    slave {
        pcm "hw:2,0"
    }
}

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: CD002 [CD002], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

 arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2019-02-09:
I guess you have not used the audio config files that i have provided. Run the USB-DAC audio installer that i have provided. Before that change the hardware ids in the  files listed inside the USB-DAC folder.

chrisf95, 2019-02-09:
I run the script USB-DAC and changed a value according to my devices and then reboot, but it's like before: I can record e play a raw file using command, but the recognition don't work, moreover every 5 min the assistant start and say hi and I can't do nothing

chrisf95, 2019-02-09:
I have to copy the .config file in root (because gave me that error) and after this the command gave me:
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-09 18:21:31,640 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[019096c0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0193d550] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication

and it's like when it autoboots and it doens't answer

shivasiddharth, 2019-02-09:
Can you run though this again ? And share the output of .asoundrc and asound.conf ?

chrisf95, 2019-02-09:
pi@raspberrypi:~ $ cat /etc/asound.conf
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:2,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
pi@raspberrypi:~ $ cat /.asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:2,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: CD002 [CD002], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

shivasiddharth, 2019-02-09:
Did you restart the Pi after altering the audio config ?

chrisf95, 2019-02-09:
Yes

shivasiddharth, 2019-02-09:
`sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]` Means that the script is not seeing your Mic.

chrisf95, 2019-02-09:
mmmh ok, but the last time it didn't give me that error (and also now that I retry)

I also try using the googlesamples-pushtotalk and it understood my voice and what I say but lock after the second "push button" without reproduce any sound

shivasiddharth, 2019-02-09:
Can you post the latest error that you are getting when you manually run the assistant?

chrisf95, 2019-02-09:
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-09 18:21:31,640 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[019096c0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0193d550] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication

I think it isn't anche error bit the assistant waiting an order

shivasiddharth, 2019-02-09:
> /home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory [...]

This is just a part of the screen. There should be a lot more after this.

chrisf95, 2019-02-09:
This Is the only terminal responsabile... It give anything

shivasiddharth, 2019-02-09:
Terminal will not stop with `Initializing GPIOs 5 and 6 for assistant activity indication`

chrisf95, 2019-02-09:
I try to reinstall in these days and give you other info

chrisf95, 2019-02-10:
I reinstall everything from a clean raspbian lite and it gave me starting manually:
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-10 03:56:07,438 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[02278a98] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[022ac928] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication

shivasiddharth, 2019-02-11:
Mic getting stuck is an upstream issue. https://github.com/googlesamples/assistant-sdk-python/issues/303 But i can say for sure that this project's pushbutton script will not stop at `Initializing GPIOs 5 and 6 for assistant activity indication`.  If you are hearing that  "Hi" then the program has moved beyond `Initializing GPIOs 5 and 6 for assistant activity indication` so obviously, the script will not end there.

chrisf95, 2019-02-11:
I see, but there's nothing also try every wakeword

bachoo786, 2019-06-12:
Did you manage to solve this?

chrisf95, 2019-06-12:
> Did you manage to solve this?

Nope, I tried different things found on internet, but it doesn't recognize the wakeword

bachoo786, 2019-06-12:
I have the same problem. I am using 2 mic Pi HAT on my raspberry pi zero w with raspbian stretch lite installed. I have installed the drivers from https://github.com/respeaker/seeed-voicecard.git which generates the /etc/asound.conf. I copy the contents of /etc/asound.conf to /home/pi/.asoundrc and it seems to work straight after installing GassistPi i.e. the first run (using custom wake word, "ok Google" doesnt work for me). After a minute or 2 it stops working, I use the custom wake word but it doesnt work and seems dead even after restarting GassistPi.

I wish there was an easy way to resolve this.

shivasiddharth, 2019-07-19:
@bachoo786 , You are not advised to use respeaker's own files. I have modified the asound.conf files to work with this project and that is why i have my own installation instructions for respeaker DACs https://github.com/shivasiddharth/GassistPi#install-audio-config-files 

I have already mentioned this a number of times. Pi Zero does not have Ok_Google wakewording.

bachoo786, 2019-07-19:
I didn't succeed with rpi zero w and moved on to orange pi zero. Installed armbian stretch but it doesn't have the drivers for audio. I then installed armbian xenial and the audio works well. Installed Google assistant following Google's instructions but the hot word library doesn't work, however pushtotalk seems to work for me. 

I understand you have mentioned it many times but is there anyway I could use snips to work with Google assistant on orange pi zero?

shivasiddharth, 2019-07-19:
@bachoo786 , GassistPI works out of the box with Orangepi Zero and Armbian stretch. You can use the audio drivers that i have given. It is not the OS that is causing the issue. Its the Kernel. Download the Armbian stretch with Linux kernel 4.14.

bachoo786, 2019-07-19:
Ok let me try that and report back.

Thanks for replying

bachoo786, 2019-07-19:
@shivasiddharth which are the audio drivers you are talking about for the orangepi zero? is it this "3.1. USB DAC or USB Sound CARD users," ? I tried "3.1. USB DAC or USB Sound CARD users," but it gives me an error when I run `root@orangepizero:~# sudo ./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh`
`./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh: line 29: /home/root/audiosetup: No such file or directory
`

shivasiddharth, 2019-07-19:
You are running it as root. The gasistpi folders will be in user directory

bachoo786, 2019-07-19:
ok ran it again without sudo and got this:

```
root@orangepizero:~# ./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh
Installing USB DAC config
root@orangepizero:~# aplay -l
aplay: device_list:270: no soundcards found...
root@orangepizero:~# speaker-test

speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
Playback open error: -2,No such file or directory

```

still throws an error and aplay -l says `aplay: device_list:270: no soundcards found...`

I am running:
```
root@orangepizero:~# uname -r
4.14.78-sunxi
```
the kernel is 4.14.78 like you told me to and its armbian stretch

shivasiddharth, 2019-07-19:
If you are using USB-DAC option you should have some card inserted otherwise change the hardware numbers in the asound files

bachoo786, 2019-07-19:
I thought in the instructions it says for non raspberry pi users use the USB-DAC option which I did. Am I meant to be using USB-DAC option for my orangepi zero?

I only have a orangepi zero with the 3.5mm additional board attach to it.

The hardware numbers are unknown as aplay-l and arecord-l give error as you saw from my previous post.

shivasiddharth, 2019-07-19:
Can you double check if the board is inserted properly, the board has to appear right.

bachoo786, 2019-07-19:
just checked everything and its fine as with a different version of armbian i can access aplay -l etc

shivasiddharth, 2019-07-19:
https://forum.armbian.com/topic/6997-aplay-l-no-soundcards-found/

bachoo786, 2019-07-19:
right that fixed the audio issue. thanks man

I will now continue with the installation.

bachoo786, 2019-07-19:
@shivasiddharth after entering the modelid I get this error:

`./GassistPi/scripts/gassist-installer.sh: line 35: /home/root/modelid.txt: No such file or directory`

Any idea on how I could resolve that?

shivasiddharth, 2019-07-20:
You are running the script as root switch to user.

bachoo786, 2019-07-20:
@shivasiddharth thats correct I made a user and ran the script it worked well until this point and I get lots of errors:

```
 Building wheel for lxml (setup.py) ... error
  ERROR: Complete output from command /home/bachoo786/env/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-in                           stall-akk0pgp6/lxml/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'                           \n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-awxlusm_ --python-tag cp35:
  ERROR: Building lxml version 4.3.4.
  Building without Cython.
  Using build configuration of libxslt 1.1.29
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/cssselect.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/__init__.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/builder.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/sax.py -> build/lib.linux-armv7l-3.5/lxml
  creating build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-3.5/lxml/includes
  creating build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/defs.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/clean.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/builder.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/html5parser.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/diff.py -> build/lib.linux-armv7l-3.5/lxml/html
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron
  copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv7l-3.5/lxml/isoschematron
  copying src/lxml/etree.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/etree_api.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/lxml.etree.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/tree.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/uri.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xmlschema.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/config.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/htmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xmlerror.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/etreepublic.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/lxml-version.h -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/etree_defs.h -> build/lib.linux-armv7l-3.5/lxml/includes
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
  copying src/lxml/isoschematron/resources/rng/iso-schematron.rng -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  copying src/lxml/isoschematron/resources/xsl/XSD2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  copying src/lxml/isoschematron/resources/xsl/RNG2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
[... 101 lines cut ...]
[... cut ...]
ERROR: Command "/home/bachoo786/env/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-akk0pgp6/lxml/se                           tup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exe                           c(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-bw2k8npi/install-record.txt --single-version-externa                           lly-managed --compile --install-headers /home/bachoo786/env/include/site/python3.5/lxml" failed with error code 1 in /tmp/pip-ins                           tall-akk0pgp6/lxml/

```

Any ideas?

shivasiddharth, 2019-07-20:
Follow #939

bachoo786, 2019-07-20:
Are you saying to increase the swap size?

bachoo786, 2019-07-21:
@shivasiddharth i followed 939 and have successfully installed it. However start from boot gives me the following error:

`Jul 21 11:22:31 orangepizero systemd[1]: gassistpi.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services. Refusing.`

shivasiddharth, 2020-03-07:
Closing this for inactivity

closed 2020-03-07 by shivasiddharth