shivasiddharth/GassistPi #823: after complete installation it work fine but after rebooting it will show error

aj6823, 2019-02-10:
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-10 23:57:19,267 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[018d9518] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[019113e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 76, in <module>
    from indicator import assistantindicator
  File "/home/pi/GassistPi/src/indicator.py", line 14, in <module>
    import numpy
  File "/home/pi/env/lib/python3.5/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
ImportError: cannot import name 'core'

closed 2019-02-10 by close-issue-app[bot]

shubham1520, 2019-04-13:
did you find a solution. i am having the same error

aj6823, 2019-04-14:
Yeah am just flush everything...update the os and start the installation again it works fine