shivasiddharth/GassistPi #830: assistance run can not run manually

aj6823, 2019-02-13:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000cb3cefc9

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Paste your Board and OS details here.   
```

### Describe the bug:
```
Assistance can not run manually it show error authentication failure
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'xxxxxxxxxxx'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-02-13 23:16:31,846 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01150cf8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01188bc0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 76, in <module>
    from indicator import assistantindicator
  File "/home/pi/GassistPi/src/indicator.py", line 14, in <module>
    import numpy
  File "/home/pi/env/lib/python3.5/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
ImportError: cannot import name 'core'

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-13:
Run the following and post the output:
```
source env/bin/activate
pip freeze
```

anshushukla2002, 2019-02-14:
same prob here

_source env/bin/activate
pip freeze_
gives

**adafruit-io==2.1
aftership==0.2
appdirs==1.4.3
asn1crypto==0.24.0
beautifulsoup4==4.7.1
bs4==0.0.1
cachetools==3.1.0
casttube==0.2.0
certifi==2018.11.29
cffi==1.11.5
chardet==3.0.4
click==6.7
cryptography==2.5
decorator==4.3.2
enum34==1.1.6
feedparser==5.2.1
future==0.17.1
futures==3.1.1
gmusicapi==12.0.0
google-api-core==1.7.0
google-api-python-client==1.7.8
google-assistant-grpc==0.2.1
google-assistant-library==1.0.1
google-assistant-sdk==0.5.1
google-auth==1.6.2
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.2.0
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.5.6
googletrans==2.3.0
gpiozero==1.4.1
gpsoauth==0.4.1
grpcio==1.18.0
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.12.0
idna==2.8
ifaddr==0.1.6
ipaddress==1.0.22
kodi-json==1.0.0
lxml==4.3.1
MechanicalSoup==0.11.0
mock==2.0.0
mutagen==1.42.0
numpy==1.16.1
oauth2client==4.1.3
oauthlib==3.0.1
pafy==0.5.4
paho-mqtt==1.4.0
pathlib2==2.3.3
pbr==5.1.2
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.6.1
psutil==5.5.0
pushbullet.py==0.11.0
pyasn1==0.4.5
pyasn1-modules==0.2.4
PyAudio==0.2.11
PyChromecast==2.4.0
pycparser==2.19
pycryptodomex==3.7.2
pyOpenSSL==19.0.0
python-dateutil==2.8.0
python-magic==0.4.15
python-vlc==3.0.4106
pytz==2018.9
PyYAML==3.13
requests==2.21.0
requests-oauthlib==1.2.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.12.0
sounddevice==0.3.12
soupsieve==1.7.3
spidev==3.2
spotipy==2.4.4
tenacity==4.12.0
uritemplate==3.0.0
urllib3==1.24.1
validictory==1.1.2
websocket-client==0.54.0
youtube-dl==2019.2.8
zeroconf==0.21.3**

shivasiddharth, 2019-02-15:
Are you running Alexa also ?

shivasiddharth, 2019-02-15:
You guys must be having a broken numpy package. Run:

```
source env/bin/activate
pip uninstall numpy
pip install numpy
```

aj6823, 2019-03-16:
**pi@raspberrypi:~ $ source env/bin/activate**
**(env) pi@raspberrypi:~ $ pip uninstall numpy**
Exception:
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 179, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/commands/uninstall.py", line 48, in run
    with self._build_session(options) as session:
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 93, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/download.py", line 344, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/download.py", line 108, in user_agent
    zip(["name", "version", "id"], distro.linux_distribution()),
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 120, in linux_distribution
    return _distro.linux_distribution(full_distribution_name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 674, in linux_distribution
    self.name() if full_distribution_name else self.id(),
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 712, in name
    name = self.os_release_attr('name') \
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 883, in os_release_attr
    return self._os_release_info.get(attribute, '')
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 550, in __get__
    ret = obj.__dict__[self._fname] = self._f(obj)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 922, in _os_release_info
    return self._parse_os_release_content(release_file)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 953, in _parse_os_release_content
    tokens = list(lexer)
  File "/usr/lib/python3.5/shlex.py", line 263, in __next__
    token = self.get_token()
  File "/usr/lib/python3.5/shlex.py", line 90, in get_token
    raw = self.read_token()
  File "/usr/lib/python3.5/shlex.py", line 118, in read_token
    nextchar = self.instream.read(1)
  File "/usr/lib/python3.5/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa5 in position 3: invalid start byte

shivasiddharth, 2019-03-16:
Check your language encoding option in Raspbian config.

aj6823, 2019-03-18:
**pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip uninstall numpy**
Exception:
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 179, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/commands/uninstall.py", line 48, in run
    with self._build_session(options) as session:
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 93, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/download.py", line 344, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/download.py", line 108, in user_agent
    zip(["name", "version", "id"], distro.linux_distribution()),
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 120, in linux_distribution
    return _distro.linux_distribution(full_distribution_name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 674, in linux_distribution
    self.name() if full_distribution_name else self.id(),
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 712, in name
    name = self.os_release_attr('name') \
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 883, in os_release_attr
    return self._os_release_info.get(attribute, '')
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 550, in __get__
    ret = obj.__dict__[self._fname] = self._f(obj)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 922, in _os_release_info
    return self._parse_os_release_content(release_file)
  File "/home/pi/env/lib/python3.5/site-packages/pip/_vendor/distro.py", line 953, in _parse_os_release_content
    tokens = list(lexer)
  File "/usr/lib/python3.5/shlex.py", line 263, in __next__
    token = self.get_token()
  File "/usr/lib/python3.5/shlex.py", line 90, in get_token
    raw = self.read_token()
  File "/usr/lib/python3.5/shlex.py", line 166, in read_token
    raise ValueError("No closing quotation")
ValueError: No closing quotation

shivasiddharth, 2019-03-19:
There seems to be some issues with your OS. Start from scratch. Closing this thread. If the issue persists, feel free to reopen.

closed 2019-03-19 by shivasiddharth

aj6823, 2019-03-26:
Can u please help me ...am final year student of cse and I want to help releated too my project...so can u please share me u r Mail id please