shivasiddharth/GassistPi #839: gmusicapi deviceid problem

Admin-HelloGroup, 2019-02-26:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000f1559930

RETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
I got an error when trying to launch GAssist Manually
gmusicapi doesn't seem to find the device id
```

```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id xxxxxxx
```

### Paste the contents of the terminal below
```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:GetDeviceManagementInfo(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#userClientIdList'}
INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 28, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined
```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-02-26:
add `print(device)` above the following line https://github.com/shivasiddharth/GassistPi/blob/cecf702f68b5a1513d09604cdefd6088af6a5295/src/mediaplayer.py#L23 and share the terminal output.

Admin-HelloGroup, 2019-02-27:
I reinstalled and re-registered the device and it's working now.

closed 2019-02-27 by shivasiddharth

Admin-HelloGroup, 2019-02-27:
Except as soon as I try to enable google music I got an error.
the device array is empty
This is very weird because the api oauth logging is True but api.get_registered_devices() doesn't return anything. I also tried logging with api.login(email, password, Mobileclient.MACADDRESS) not working neither... Let me know if you find a fix to this Thank you

shivasiddharth, 2019-02-27:
Open Google Play Music in a browser, go into settings and check the list of devices and get back.

samail1995, 2019-03-21:
i did try everything to fix can you help with this plz

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 29, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined