shivasiddharth/GassistPi #842: Bluetooth Speaker and USB Microphone not working at all with fresh stretch install

bachoo786, 2019-03-02:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```
On a PI3 B+ I cloned https://github.com/shivasiddharth/GassistPi at the last step it throws an error.

Google Assistant doesnt work at all for me.

I suspect its something to do with my audio. I am using USB mic which is already connected to my rpi and bluetooth speaker which also already connected and working fine. I checked this as I am able to play music through my speaker via bluetooth as well as open youtube in the browser and play video.

I have got no idea on what to do and it is abit frustrating.

This works.
aplay -D bluealsa:DEV="20:18:08:30:A9:66" /usr/share/sounds/alsa/hello.wav

........

Here is my ~/.asoundrc and my /etc/asound.conf. Please note that these 2 files are the same.

pcm.dsnooper {
type dsnoop
ipc_key 816357492
ipc_key_add_uid 0
ipc_perm 0666
slave {
pcm "hw:1,0"
channels 1
}
}

pcm.bluetooth {
type plug
slave.pcm {
type bluealsa
device "20:18:08:30:A9:66"
profile "a2dp"
}
}

pcm.!default {
type asym
playback.pcm {
type plug
slave.pcm "bluetooth"
}
capture.pcm {
type plug
slave.pcm "dsnooper"
}
}
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes
[... 120 lines cut ...]
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 1024, in 
Myassistant().main()
File "/home/pi/GassistPi/src/main.py", line 192, in init
self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in init
stream_callback=audio_callback)
File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
stream = Stream(self, *args, **kwargs)
File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in init
self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 1026, in 
logger.exception(error)
NameError: name 'logger' is not defined
```

### Attach the log file   
Log file can be found in /tmp/ directory  
There is no log file.

shivasiddharth, 2019-03-02:
Bluetooth audio is not a part of the project and i do not support issues related to bluetooth audio.

bachoo786, 2019-03-02:
why do you have to close the topic? can you not help? I have been trying my utmost to get this working. 
and how is it a bluetooth audio issue?

bachoo786, 2019-03-02:
@shivasiddharth there is no point of creating this and not willing to help others, your stupid one line answers and closing of topics dont do any good.

if you are not willing to help then FUCK OFF

shivasiddharth, 2019-03-02:
Use of unparliamentary language is uncalled for. If dont go through the README, its not my fault. https://github.com/shivasiddharth/GassistPi#do-not-raise-an-issue-request-for-non-issue-stuff-for-non-issue-help-and-interaction-use-gitter-

closed 2019-03-02 by shivasiddharth

bachoo786, 2019-03-02:
Oh the irony you have helped others with bluetooth speakers. Maybe not your game. Apologies anyway