shivasiddharth/GassistPi #849: When playing music from youtube I always get the same song

igiul667, 2019-03-15:
Hi, I installed gassistpi yesterday, since google assistant on rpi couldn't play music on it's own. I have a problem when trying to play a song from youtube. It always plays the same song no matter what I say. I already checked [https://github.com/shivasiddharth/GassistPi/issues/753](url) and i tried 
pip install -r /home/pi/GassistPi/Requirements/GassistPi-pip-requirements.txt
as suggested in the post but I still get the same issue. Have a nice day

closed 2019-03-15 by close-issue-app[bot]

aandroide, 2019-03-15:
surely you're wrong to say the command. the correct command is: ok google, play "song" from youtube

igiul667, 2019-03-15:
I'm shure the command is right, I'm actually italian but it shoud be a supported language. Also should I open a new issue or keep using this?

aandroide, 2019-03-15:
Se  nel file config.yaml hai impostato it-it,  il comando da dire in italiano è: riproduci bla bla bla da youtube

igiul667, 2019-03-15:
Yes it is set to it-IT

aandroide, 2019-03-15:
Hai anche inserito l'api key di youtube corretamente? Ed hai controllato che sia realmente attiva nella tua consolle di google?

igiul667, 2019-03-15:
Ho appena finito di reinstallare ed ora va, grazie mille della disponibilità e buona giornata