shivasiddharth/GassistPi #850: Domoticz server not online error

pimseb, 2019-03-16:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000d2ec91e2

```

### Describe the bug:
```
Everything worked fine until I enabled Domoticz in config.yaml
Gassispi thinks the server is not online but it's in fact online and it's also good set up in the config file 
Domoticz:
  # Set this to 'Enabled' to control devices registered to Domoticz server.
  Domoticz_Control: 'Disabled'
  Server_IP:
    # If you are running the server on the Pi itself, then change the ip to 'localhost'
    - '192.168.1.20'
  Server_port:
[... 151 lines cut ...]
Mar 16 19:53:59 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=11/SEGV
Mar 16 19:53:59 raspberrypi systemd[1]: gassistpi.service: Unit entered failed state.
Mar 16 19:53:59 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 16 19:53:59 raspberrypi systemd[1]: gassistpi.service: Service hold-off time over, scheduling restart.
Mar 16 19:53:59 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 16 19:53:59 raspberrypi systemd[1]: Started Google Assistant-Voice.
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/2974461/GassistPi.log)

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-03-16:
The port number is wrong. it should be 443.

pimseb, 2019-03-18:
Will it also work if I use another https port than 443 ?

DewGew, 2019-03-19:
Domoticz is using port 443 if you didnt change that during Domoticz setup

shivasiddharth, 2020-03-07:
Closing this for inactivity

closed 2020-03-07 by shivasiddharth