shivasiddharth/GassistPi #853: High CPU usage for AudioInputProce(ss?)

gartensofa, 2019-03-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
processor : 0 model name : ARMv7 Processor rev 5 (v7l) BogoMIPS : 64.00 Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm CPU implementer : 0x41 CPU architecture: 7 CPU variant : 0x0 CPU part : 0xc07 CPU revision : 5 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)" NAME="Raspbian GNU/Linux" VERSION_ID="9" VERSION="9 (stretch)" ID=raspbian ID_LIKE=debian HOME_URL="http://www.raspbian.org/" SUPPORT_URL="http://www.raspbian.org/RaspbianForums" BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Hey Guys,
i noticed a high cpu load for the "python"-pcocess, so i decided to investigate.
The process has an idle load of around 40-50%. With "top -H" i can see that the
"AudioInputProce" takes about 20-30%.
I'm using an USB microphone and everything works flawlessly, but for the high CPU usage..
I also got kodi running over the optical output of a HiFiberry card, but stopping kodi makes no difference..
Any input on this would be highly appreciated!
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
none
```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

shivasiddharth, 2019-03-19:
Have you used the audio config files that i have provided or some others ?

gartensofa, 2019-03-19:
Hey, thanks for the quick reply.
I got both asound-files just like the provided ones.

shivasiddharth, 2019-03-19:
Can you post them here?

gartensofa, 2019-03-19:
Sure:
/etc/asound.conf
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

/home/pi/.asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

shivasiddharth, 2019-03-20:
You need to do the following for me to get a grip of what is going on.
1. Check the CPU usage with Kodi control enabled and disabled in config.yaml.
2. Use the USB-DAC audio config that that i provided and see if it has any effect.
3. Share the snapshot of the process/CPU usage tree.

gartensofa, 2019-03-20:
1. Dis/enabling kodi control makes no meassurable difference.
2. i exchanges both asound files and rebooted, still the same.
3. here is a link to a screenshot of "top -H", i hope thats what you meant. Otherwise please instruct me further. I am much more eager than i am competent ;-)
http://imgur.com/a/mtDTC1w

shivasiddharth, 2019-03-20:
As far as i know the AudioInputprocess has nothing to do with GassistPi (could be wrong).  Even if you stop the GassistPi process the Audioinput will linger around if what i say i correct. Please do double check that.
Deactivating the pushbutton thread will reduce the CPU usage considerably. 

https://github.com/shivasiddharth/GassistPi/blob/cecf702f68b5a1513d09604cdefd6088af6a5295/src/main.py#L302 

either comment out that line or change the "start" part to "stop".

shivasiddharth, 2019-03-20:
Before opening a ticket there let's see if it's got something to do with this project. Do you have custom wakeword enabled? Try disabling that in the config.yaml file

gartensofa, 2019-03-21:
Interesting. That brought the main python pcocess down to around 35%!
..but the audioinput-thread is not affected by it..

gartensofa, 2019-03-21:
Now i started the assistant with
source env/bin/activate
googlesamples-assistant-hotword --device-model-id <deviceid>
And the same audioinput-thread with the same cpu usage came up. I guess that means it's a problem within the sdk?
Do you think i could bypass it with custom wakeword? Is it possible to completely disable the google listening process with costum wakeword?

shivasiddharth, 2019-03-21:
30-35% is pretty normal for Google assistant. You cannot go below that. Opening an issue with SDK is not going to help either, because this CPU utilization is not an anomaly.

shivasiddharth, 2019-03-21:
In the issue template your hardware id is missing. May i know as to what board are you using?

gartensofa, 2019-03-21:
Sure, sorry. Its an a01041, so its a rpi2 model b 1.1 with 1gig ram, 
of wich 320M are split towards the gpu.

shivasiddharth, 2019-03-21:
1. Is the CPU usage that you are reporting for a manually started assistant or the assistant running as service?
2. Can you run it as a service on start and revert back with results?

gartensofa, 2019-03-21:
1. That last one was from starting it manually
2. Could you tell me how to run it as service?

shivasiddharth, 2019-03-21:
https://github.com/shivasiddharth/GassistPi#headless-autostart-on-boot-service-setup

gartensofa, 2019-03-21:
Oh, sorry. I misunderstood. All of the above is relating to starting gassistpi as a service.
(except for https://github.com/shivasiddharth/GassistPi/issues/853#issuecomment-475158118 when i started the google sample directly and had the same CPU usage problem.)
Here is what happens when i start gassistpi manually:

`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id <censored>
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-03-21 22:50:47,156 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01850950] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[018528a0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
[... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: gassistpi-3b899-gassistpi-75mleh
device_id: <censored>

    This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE
`

shivasiddharth, 2019-03-26:
30-35% is typical of the google assistant. Unfortunately there is nothing much more that i can do.

shivasiddharth, 2020-03-07:
Closing this for inactivity

closed 2020-03-07 by shivasiddharth