shivasiddharth/GassistPi #854: can not install GassistPi on hassbian Stretch

hoatienii, 2019-03-20:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file   
Log file can be found in /tmp/ directory

hoatienii, 2019-03-20:
I install GassistPi on hassbian, it get error:

_Using cached https://files.pythonhosted.org/packages/07/ca/bc827c5e55918ad223d59d299fff92f3563476c3b00d0a9157d9c0217449/cryptography-2.6.1.tar.gz
  Installing build dependencies ... error
  Complete output from command /home/pi/env/bin/python /home/pi/env/lib/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-tbrqhrao/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=18.5 wheel "cffi>=1.8,!=1.11.3; python_implementation != 'PyPy'":
  Traceback (most recent call last):
    File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/home/pi/env/lib/python3.6/site-packages/pip/__main__.py", line 16, in <module>
      from pip._internal import main as _main  # isort:skip # noqa
    File "/home/pi/env/lib/python3.6/site-packages/pip/_internal/__init__.py", line 4, in <module>
      import locale
    File "/usr/local/lib/python3.6/locale.py", line 16, in <module>
      import re
    File "/usr/local/lib/python3.6/re.py", line 142, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'

  ----------------------------------------
Command "/home/pi/env/bin/python /home/pi/env/lib/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-tbrqhrao/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=18.5 wheel "cffi>=1.8,!=1.11.3; python_implementation != 'PyPy'"" failed with error code 1 in None_

shivasiddharth, 2019-03-20:
These are the only OSes that i support now. https://github.com/shivasiddharth/GassistPi#only-oses-suported-are

closed 2019-03-20 by shivasiddharth