shivasiddharth/GassistPi #866: Getting No Audio/Mic Inputs, Nothing. (USB MIC + Audio Jack Speakers)

GamingFlamin, 2019-04-15:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```

```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 89.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000b71a0519

```
```
Paste your Board and OS details here.   

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
[... 123 lines cut ...]
Not a single log at all.
Looked all over the /tmp/
only directories.
No mentions of Gassistpi.log

Pulled from systemctl's log 

```
Apr 15 16:59:31 raspberrypi python[1395]: INFO:gmusicapi.Mobileclient1:logged out
Apr 15 16:59:31 raspberrypi python[1395]: Traceback (most recent call last):
Apr 15 16:59:31 raspberrypi python[1395]:   File "/home/pi/GassistPi/src/main.py", line 51, in <module>
Apr 15 16:59:31 raspberrypi python[1395]:     from actions import say
Apr 15 16:59:31 raspberrypi python[1395]:   File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
Apr 15 16:59:31 raspberrypi python[1395]:     from mediaplayer import api
Apr 15 16:59:31 raspberrypi python[1395]:   File "/home/pi/GassistPi/src/mediaplayer.py", line 28, in <module>
Apr 15 16:59:31 raspberrypi python[1395]:     logged_in=api.oauth_login(deviceid)
Apr 15 16:59:31 raspberrypi python[1395]: NameError: name 'deviceid' is not defined
Apr 15 16:59:32 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Apr 15 16:59:32 raspberrypi systemd[1]: gassistpi.service: Unit entered failed state.
Apr 15 16:59:32 raspberrypi systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Apr 15 16:59:32 raspberrypi systemd[1]: gassistpi.service: Service hold-off time over, scheduling restart.
Apr 15 16:59:32 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Apr 15 16:59:32 raspberrypi systemd[1]: Started Google Assistant-Voice.

```

GamingFlamin, 2019-04-15:
Just to point out, Sorry if this is a simple issue, I've just can't seem to fix it myself.

shivasiddharth, 2019-04-16:
> INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 28, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined

This has been discussed here in #839 . Regarding the mic not responding, not all mics work with the google assistant. If you are having one of those adafruit's $5 USB mic, try changing the mic.

GamingFlamin, 2019-04-16:
Here's the issue, this has worked before with this mic, even the old google assistant method worked before, So I know this isn't a USB Mic issue, I can simply go back to the other ways of putting the google assistant on it, but then again, I don't get any of the features that GassistPI

The mic in question is a Blue Snowball USB Mic, not a "$5 mic"

shivasiddharth, 2019-04-16:
1. Are you getting the bootup chime in a loop (continuously)?
2. Post the outputs of the following:
```
  sudo nano /etc/asound.conf
  sudo nano /.asoundrc
  aplay -l
  arecord -l
```

GamingFlamin, 2019-04-16:
1. The bootup chime happened randomly once, I guess the PI rebooted but then again, I had no logs of it rebooting.. I did another reboot manually, and got no chime then.
2. 
  sudo nano /etc/asound.conf
 ```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

```

Nothing in   sudo nano /.asoundrc

```
pi@raspberrypi:~$   aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 6/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

```
pi@raspberrypi:~$   arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Snowball [Blue Snowball], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

shivasiddharth, 2019-04-16:
You are not getting any boot up chime because of the gmusic error (https://github.com/shivasiddharth/GassistPi/issues/866#issuecomment-483589892). That needs to be resolved first. Copy the .asoundrc file that i have given and place it in the /home/pi/ directory

GamingFlamin, 2019-04-16:
Hold up, what do you mean revert back, the only steps I see is just adding 
`add print(device) above the following line`

shivasiddharth, 2019-04-16:
There two test points https://github.com/shivasiddharth/GassistPi/issues/839#issuecomment-467458554
https://github.com/shivasiddharth/GassistPi/issues/839#issuecomment-467897860 
By revert back i meant return to me with the results (If that confused you).

Files prefixed with "." will be hidden in pi. Right click in the folder and choose to view hidden files.

Mate, i have attended to each every issue in this git and i know what relates to what. Its up to you to either trust me or disregard my instructions.

shivasiddharth, 2019-04-16:
1. After adding the print device, manually run the assistant. Thats what will print the stuff on the terminal.
2. You can either create a new file in the /home/pi/ directory and paste the contents or as it is copy and paste the file from the "/GassistPi/audio-drivers/USB-MIC-JACK/scripts/".

GamingFlamin, 2019-04-16:
Thanks for clarifying 
1. 
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 24
    if device['type']=='ANDROID':
                                ^
IndentationError: unindent does not match any outer indentation level
pi@raspberrypi:~$
```
2. I'll do that now

GamingFlamin, 2019-04-16:
I have now added the .asoundrc into /home/pi

shivasiddharth, 2019-04-16:
> Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 24
    if device['type']=='ANDROID':
                                ^
IndentationError: unindent does not match any outer indentation level
pi@raspberrypi:~$

Looks like you have used a TAB. You should move the "if" commands around only with SPACE.

GamingFlamin, 2019-04-16:
Yep, sorry, I found the issue

GamingFlamin, 2019-04-16:
Here's the console output.
```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#configList', 'data': {'entries': [{'key': 'canManageFP', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'shouldShowLocalSuggestions', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'purchaseAccess', 'kind': 'sj#configEntry', 'value': '3'}, {'key': 'wsVdtMillis', 'kind': 'sj#configEntry', 'value': '30000'}, {'key': 'sideloadedPlaybackLogging', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isWarmWelcomeDisabledForWoodstock', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'canCancelSubscription', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'nashvillebg', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'isVsAvailable', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'canBrowseSt', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'appstoreRatingPrompt', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'upsellDrawerCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE NOW'}, {'key': 'upsellDrawerDownloadMessage', 'kind': 'sj#configEntry', 'value': "Download and listen to any song, album, or station when you're not connected"}, {'key': 'wsVatMillis', 'kind': 'sj#configEntry', 'value': '5000'}, {'key': 'nuqRequiredForWsSignup', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'upsellSkipLimitTwoLeftCtaMessage', 'kind': 'sj#configEntry', 'value': 'GET MORE'}, {'key': 'upsellNewReleasesCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'upsellListenNowCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'enableWsExtras', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'upsellSignedOutMessage', 'kind': 'sj#configEntry', 'value': 'Subscribe to get millions of songs on-demand, offline, and ad-free'}, {'key': 'topListMaxSize', 'kind': 'sj#configEntry', 'value': '500'}, {'key': 'upsellSkipLimitTwoLeftMessage', 'kind': 'sj#configEntry', 'value': '2 skips left'}, {'key': 'showLMark', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'conciergeListenNow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'minimumAppStartCountForFreeUpsell', 'kind': 'sj#configEntry', 'value': '6'}, {'key': 'upsellPreviousCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'upsellDownloadMessage', 'kind': 'sj#configEntry', 'value': "Download and listen to any song, album, or station when you're not connected"}, {'key': 'showYoutubeRedInfoCard', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'concListenNow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'instantSearchDelay', 'kind': 'sj#configEntry', 'value': '300'}, {'key': 'upsellChartsMessage', 'kind': 'sj#configEntry', 'value': 'Subscribe to play any song or album'}, {'key': 'isPodcastSyncEnabled', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'upsellChartsCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'showViOSL', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'topListUpdatePeriodSeconds', 'kind': 'sj#configEntry', 'value': '604800'}, {'key': 'isSuggestDeletionEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'shouldShowServerSuggestions', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'situationsResponseTTLMinutes', 'kind': 'sj#configEntry', 'value': '30'}, {'key': 'isFoplessNautilusUser', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'appStartCountForFreeUpsellRevisionNumber', 'kind': 'sj#configEntry', 'value': '1459804296734'}, {'key': 'mgRPBDesc', 'kind': 'sj#configEntry', 'value': ''}, {'key': 'partridgeManagement', 'kind': 'sj#configEntry', 'value': '2'}, {'key': 'showB', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'showA', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'shouldEnableThriller', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'isPodcastsEnabledZ', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'upsellThumbsUpMessage', 'kind': 'sj#configEntry', 'value': 'Subscribe to play any song or album'}, {'key': 'topListPageSize', 'kind': 'sj#configEntry', 'value': '100'}, {'key': 'isHallmarkEnabled', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'upsellThumbsUpCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'uploadAccess', 'kind': 'sj#configEntry', 'value': '2'}, {'key': 'isNautilusAvailable', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'nautilusAccess', 'kind': 'sj#configEntry', 'value': '2'}, {'key': 'isQuizEnabledForWoodstockOnSignup', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'showO', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'disabledConnectivityTimeoutMs', 'kind': 'sj#configEntry', 'value': '200000'}, {'key': 'hasPControlSettings', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'wasRecentlyInFoplessNautilus', 'kind': 'sj#configEntry', 'value': 'false'}, {'key': 'showV', 'kind': 'sj#configEntry', 'value': 'true'}, {'key': 'upsellSignedOutCtaMessage', 'kind': 'sj#configEntry', 'value': 'SUBSCRIBE'}, {'key': 'forceSignupFlow', 'kind': 'sj#configEntry', 'value': 'true'}, {'key' [... cut ...]
DEBUG:gmusicapi.Mobileclient1:GetDeviceManagementInfo(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#userClientIdList'}
INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 29, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined

```

shivasiddharth, 2019-04-16:
I dont see the outputs from the print(device) at all here. Have you gone through the google music authorization step ?

GamingFlamin, 2019-04-16:
I did, I did it over again to make sure. 
`
pi@raspberrypi:~/GassistPi/src$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/Extras/gmusicauth.py

Visit the following url:
 (url)
Follow the prompts, then paste the auth code here and hit enter: (code)
`

shivasiddharth, 2019-04-16:
For the gmusic api to work, the authorized device should appear up in the web version of google play music. It is surprising that you don't have any device at all listed there. If you have gmusic app installed in any of your smart device (apple or android), at least that should appear there.
For time being to test the mic, comment out these two lines https://github.com/shivasiddharth/GassistPi/blob/10bff4c2382d8a06212a26de81b9a85648634660/src/mediaplayer.py#L27-L28 . But if you use the gmusic feature, you may run into this issue https://github.com/shivasiddharth/GassistPi/issues/818#issuecomment-462406398

GamingFlamin, 2019-04-16:
Commenting out those two lines and testing,
Result: `
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#userClientIdList'}
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
[... cut ...]
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                 k
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                 k
device_model_id: raspbberypi2-test-qyq54w
device_id: 33DA287BD0DF0D274F3F5CD38333F7AE

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED
`

GamingFlamin, 2019-04-16:
Commented out two lines.
Result:

```
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#userClientIdList'}
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                 35_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                 or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                 ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
[... cut ...]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                 k
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: 
device_id: 

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

```

shivasiddharth, 2019-04-16:
Check if you have the .asoundrc sill in the /home/pi/ directory. Before starting the google assistant, plug out the mic and plug it in back again and then try starting the assistant.

GamingFlamin, 2019-04-16:
Still getting nothing from the microphone.

GamingFlamin, 2019-04-16:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

```
The file is in the /home/pi directory.

shivasiddharth, 2019-04-16:
Test your mic using:
```
arecord --format=S16_LE --duration=5 --rate=16000 --file-type=raw out.raw
aplay --format=S16_LE --rate=16000 out.raw
```

shivasiddharth, 2019-04-16:
Also in the .asoundrc and asound.conf, try changing the channels value from 1 to 2

GamingFlamin, 2019-04-16:
It's working now, I followed some steps on https://stackoverflow.com/questions/23975046/raspberry-pi-is-not-recording-from-usb-microphone#35063183

Is there anything else I need to do? Doesn't seem to be working when it first booted up. 
Hey google commands are working.
```

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

GamingFlamin, 2019-04-16:
Noteable, I'm getting an issue with youtube streaming
`
```
ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py                                                ", line 1232, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py                                                ", line 1140, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py                                                ", line 1207, in _parse_sig_js
    initial_function = jsi.extract_function(funcname)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/jsinterp.py", line 2                                                45, in extract_function
    raise ExtractorError('Could not find JS function %r' % funcname)
youtube_dl.utils.ExtractorError: Could not find JS function 'encodeURIComponent'                                                ; please report this issue on https://yt-dl.org/bug . Make sure you are using th                                                e latest version; see  https://yt-dl.org/update  on how to update. Be sure to ca                                                ll youtube-dl with the --verbose flag and include its complete output.
 (caused by ExtractorError("Could not find JS function 'encodeURIComponent'; ple                                                ase report this issue on https://yt-dl.org/bug . Make sure you are using the lat                                                est version; see  https://yt-dl.org/update  on how to update. Be sure to call yo                                                utube-dl with the --verbose flag and include its complete output.",)); please re                                                port this issue on https://yt-dl.org/bug . Make sure you are using the latest ve                                                rsion; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-                                                dl with the --verbose flag and include its complete output.
```
`

shivasiddharth, 2019-04-26:
Your youtube_dl is outdated check https://github.com/shivasiddharth/GassistPi/issues/521#issuecomment-420043868

shivasiddharth, 2020-11-02:
Closing this for inactivity.

closed 2020-11-02 by shivasiddharth