shivasiddharth/GassistPi #867: Installation error cryptography

datawire1337, 2019-04-17:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
**cat /proc/cpuinfo**  
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 0000000098305203

**cat /etc/os-release**

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
Upon installation I get an error which is pasted below. 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Collecting cryptography>=1.3.4; extra == "secure" (from urllib3[secure]<2,>=1.21; extra == "samples"->google-assistant-sdk[samples]==0.5.1)
  Using cached https://files.pythonhosted.org/packages/07/ca/bc827c5e55918ad223d59d299fff92f3563476c3b00d0a9157d9c0217449/cryptography-2.6.1.tar.gz
  Installing build dependencies ... error
  Complete output from command /home/pi/env/bin/python /home/pi/env/lib/python3.5/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-uk5qftv7/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=18.5 wheel "cffi>=1.8,!=1.11.3; python_implementation != 'PyPy'":
  Traceback (most recent call last):
    File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/home/pi/env/lib/python3.5/site-packages/pip/__main__.py", line 16, in <module>
      from pip._internal import main as _main  # isort:skip # noqa
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/__init__.py", line 40, in <module>
      from pip._internal.cli.autocompletion import autocomplete
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/autocompletion.py", line 8, in <module>
      from pip._internal.cli.main_parser import create_main_parser
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/main_parser.py", line 12, in <module>
      from pip._internal.commands import (
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/commands/__init__.py", line 6, in <module>
      from pip._internal.commands.completion import CompletionCommand
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/commands/completion.py", line 6, in <module>
      from pip._internal.cli.base_command import Command
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 25, in <module>
      from pip._internal.index import PackageFinder
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/index.py", line 41, in <module>
      from pip._internal.wheel import Wheel
    File "/home/pi/env/lib/python3.5/site-packages/pip/_internal/wheel.py", line 7, in <module>
      import compileall
    File "/usr/lib/python3.5/compileall.py", line 20, in <module>
      from concurrent.futures import ProcessPoolExecutor
    File "/home/pi/env/lib/python3.5/site-packages/concurrent/futures/__init__.py", line 8, in <module>
      from concurrent.futures._base import (FIRST_COMPLETED,
    File "/home/pi/env/lib/python3.5/site-packages/concurrent/futures/_base.py", line 381
      raise exception_type, self._exception, self._traceback
                          ^
  SyntaxError: invalid syntax

  ----------------------------------------
Command "/home/pi/env/bin/python /home/pi/env/lib/python3.5/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-uk5qftv7/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools>=18.5 wheel "cffi>=1.8,!=1.11.3; python_implementation != 'PyPy'"" failed with error code 1 in None

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-04-17:
The terminal contents are truncated. Can you provide the full terminal contents? Is this Pi's Raspbian or Hassbian ?

datawire1337, 2019-04-17:
Hello,

What do you mean exactly?
It's Raspbian.

datawire1337, 2019-04-17:
I've done a new and fresh Raspbian install and it's the same result.
I'll try this with another RPi 3B+ and get back to you aswell. 

Which terminal content do you need?

shivasiddharth, 2019-04-17:
Run the following in the terminal and re run the installer.
```
source env/bin/activate
pip uninstall futures
```

datawire1337, 2019-04-17:
I did this, and it gives me the same error.

datawire1337, 2019-04-17:
I've done like 10 installations in the past and they were all successful. 
I don't see what I'm doing different, just following the steps from your README like always. :o

poulsp, 2019-04-17:
I have the same error right now.
 RPi 3B+
2019-04-08-raspbian-stretch-lite

shivasiddharth, 2019-04-17:
I am away from my pi, so can only suggest stuff to try. Try this and then run the installer.
```
source env/bin/activate
pip install cryptography==2.4.2
```

poulsp, 2019-04-17:
The workaround
  source env/bin/activate
  pip install cryptography==2.4.2
Works.
Thank you :)

shivasiddharth, 2019-04-17:
I have updated the master. If the issue persists please feel free to reopen.

closed 2019-04-17 by shivasiddharth

datawire1337, 2019-04-18:
This worked indeed!
Tested it with the update from master aswell.

Thanks!

engperini, 2022-11-04:
Please, same error. 

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: d03114
Serial		: 100000007761dd53
Model		: Raspberry Pi 4 Model B Rev 1.4

pi@raspberrypi:~ $ cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Describe the bug:
Upon installation I get an error which is pasted below. 
Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

2. Is the issue related to assistant installation ?
Yes

If answer to question 2 is Yes, then paste the contents of the terminal below.

[... 355 lines cut ...]
          pip: n/a
          setuptools: 65.5.0
          setuptools_rust: 1.5.2
          rustc: n/a
          =============================DEBUG ASSISTANCE=============================
      
      error: can't find Rust compiler
      
      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
      
      To update pip, run:
      
          pip install --upgrade pip
      
      and then retry package installation.
      
      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
      
      This package requires Rust >=1.48.0.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

commit e8d65ecd (2019-04-17): Update GassistPi-pip-requirements.txt

Fixes #867