shivasiddharth/GassistPi #874: Assistant does not start

gopalasubramanium, 2019-04-19:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
pi@SmartScreen:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000095f34cbc
pi@SmartScreen:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

[... 119 lines cut ...]
  File "/home/pi/GassistPi/src/main.py", line 192, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9999] Unanticipated host error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1026, in <module>
    logger.exception(error)
NameError: name 'logger' is not defined

```

### Attach the log file named Gassistpi.

log
Log file can be found in /tmp/ directory  

File is empty.

shivasiddharth, 2019-04-19:
1. Post the outputs of the following:
```
  sudo nano /etc/asound.conf
  sudo nano /.asoundrc
  aplay -l
  arecord -l
```
2. Share the contents of the config.yaml from this section.

gopalasubramanium, 2019-04-19:
**1**
**sudo nano /etc/asound.conf**

pi@SmartScreen:~ $ sudo nano /etc/asound.conf
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

**sudo nano /.asoundrc**
pi@SmartScreen:~ $ sudo nano /.asoundrc
No Output

**aplay -l**
pi@SmartScreen:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: G1 [Sound BlasterX G1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**arecord -l**
pi@SmartScreen:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: G1 [Sound BlasterX G1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**2. Share the contents of the config.yaml from this section**

#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'
#    - '/home/pi/GassistPi/src/resources/smart_mirror.umdl'
#    - '/home/pi/GassistPi/src/resources/alexa.umdl'

shivasiddharth, 2019-04-19:
Run:
`sudo nano /.asoundrc`
and paste the contents of asound.conf, save and exit. And then give it a try.

gopalasubramanium, 2019-04-19:
Done but no change sir, still the same error. It works just fine when I run this "sudo  ./GassistPi/scripts/gassist-installer.sh " and enter the required details.

shivasiddharth, 2019-04-19:
This is a snowboy issue has nothing to do with Google Assistant. Perhaps an incompatible mic can you try changing the mic?

gopalasubramanium, 2019-04-20:
This solved the problem, t was because pulseaudio driver was broken and/or permissions not set to root.

`# clean and reinstall pulseaudio
sudo apt-get remove --purge alsa-base pulseaudio
sudo apt-get install alsa-base pulseaudio
sudo apt-get -f install && sudo apt-get -y autoremove && sudo apt-get autoclean && sudo apt-get clean && sudo sync && echo 3 | sudo tee /proc/sys/vm/drop_caches
# fixes user folder permissions
sudo chown -R $USER:$USER $HOME/
# then reboot
sudo shutdown -r -t 0`

Thanks a lot :)

shivasiddharth, 2019-04-26:
There was no need to install pulseaudio in the first place. But glad that you figured out the issue.

closed 2019-04-26 by shivasiddharth

makspouzer, 2019-07-19:
I am also facing same issue, I tried pulse audio reinstall and folder permission.. still no solution..

makspouzer, 2019-07-19:
(env) root@raspberrypi:/home/pi# /home/pi/GassistPi/src/main.py --device_model_id makpisep-1-makpisep-1-7hzho1
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.0.0
INFO:gmusicapi.Mobileclient1:logging to: /root/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /root/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-07-20 00:23:22,992 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/root/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/root/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
[029492f8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[02953588] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Respeaker-4-Mic
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Fatal Python error: Aborted

Thread 0x6e991470 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/queue.py", line 164 in get
  File "/home/pi/GassistPi/src/indicator.py", line 200 in _run
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x76f60640 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 98 in __init__
  File "/home/pi/GassistPi/src/main.py", line 988 in main
  File "/home/pi/GassistPi/src/main.py", line 1025 in <module>
Aborted

shivasiddharth, 2019-07-19:
@makspouzer Are you running this on buster ?

makspouzer, 2019-07-19:
I had it already installed in pi3 B, it was working perfect... I did apt-get update and apt-get upgrade. then it stopped working. So, I tried to reinstall and started seeing this problem.

makspouzer, 2019-07-19:
After installation, at the end google assistant worked fine without issue.. but main.py doesn't work

shivasiddharth, 2019-07-19:
Google assistant does not work with Buster. Check the issues already reported on this. If you are using Pi 3B, use only stretch and do not upgrade.

makspouzer, 2019-07-19:
I am not sure what buster is ? 

If you are using Pi 3B, use only stretch --> could you please point which one ?

shivasiddharth, 2019-07-19:
Latest version of Raspbian OS is called Buster. The previous version of Raspbian is called Stretch. You can download older versions in the raspberry website. Do not run sudo apt-get upgrade.

makspouzer, 2019-07-19:
Thanks.. but If I invoke google assistant separately it is working fine for me.. only if I run main.py under gassist pi it fails.

makspouzer, 2019-07-19:
Fixed. restarting pulseaudio made it work.
systemctl --user restart pulseaudio