shivasiddharth/GassistPi #875: Google can hear me, but no sound on replyfrom it, although text shows its replying

mosesgothi, 2019-04-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000003f3e898b
  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
  
```
``` 
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3101527/GassistPi.log)

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "can you hear me"}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_RENDER_RESPONSE:
  {"text": "Loud and clear", "type": 0}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-04-22:
Check point 6 under https://github.com/shivasiddharth/GassistPi#continue-after-setting-up-audio

shivasiddharth, 2019-05-31:
Closing this for inactivity.

closed 2019-05-31 by shivasiddharth