shivasiddharth/GassistPi #879: Disabling Ok Google mutes the mic, have to reset mic_state every time

shankargopal, 2019-04-24:
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000098fd4098
```

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
Go to `config.yaml` and set `Ok_Google: 'Disabled'`.  Custom_Wakeword is Enabled.

Restart with `sudo systemctl restart gassistpi.service`

The assistant comes up, says 'Hi', and then announces "Your mic has been muted."
[... 137 lines cut ...]
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3111086/GassistPi.log)

shivasiddharth, 2019-04-24:
This is not a bug but a feature. Even if the mic is muted, it will actively lookout for the custom wakeword if it was enabled. If you unmute the mic, it will start listening for the Ok-Google wakeword, which kind of kills the purpose of disabling it in the config.yaml.

shankargopal, 2019-04-24:
I see :smiley:  The annoucement that "your mic is muted" led me to not try the custom wakeword as well. However, when I try it now, I find that you're right and it picks it up. But any action that requires the assistant - for instance, "what's the time" - does not get a response.  Only custom actions that are local (for instance, "play xyz with youtube") work.  Is that also intentional?

shankargopal, 2019-04-24:
I see :smiley:.  You're right - that's indeed a feature!

Apologies for the last comment - the problem i was reporting there seems to have vanished on its own (?) possibly due to a volume command. 

The only remaining issue is that it announces that the "mic is on" and the "mic is muted" every time. Is there a way to stop it doing that?  Should I open a separate issue for that?

shivasiddharth, 2019-04-24:
There is no way to turn that off. Its from google's cloud. Can be annoying but thats the trade-off for privacy and muting the mic.

shankargopal, 2019-04-24:
Ok, thanks again for all the hard work.  Perhaps a note can be put in config.yaml that this will happen, so that users don't get confused by these "your mic is off" announcements? 

Please close the issue if you think that's not necessary.  Many thanks again. My children are going to love this :smiley:

closed 2019-04-25 by shankargopal