shivasiddharth/GassistPi #880: Include updating of youtube-dl in update.sh

shankargopal, 2019-04-24:
Youtube-dl has to update very frequently to keep up with changes in Youtube. Today (Apr 24 2019) for instance, youtube-dl broke just one week after the last update. I was able to fix this in GassistPi by the following:

```
cd /home/${USER}
source env/bin/activate
pip install --upgrade youtube-dl
```

Many users will not know how to do this, so would suggest including this in the `update.sh` script.

closed 2019-04-24 by close-issue-app[bot]

shivasiddharth, 2019-04-24:
Does the assistant crash for outdated youtube_dl ? It was supposed to raise a voice prompt that something went wrong.

shankargopal, 2019-04-24:
It doesn't crash - it handles it gracefully by saying "There was an exception, please check the logs" (or something to that effect).  But again many users may find the exception in the logs and not know what to do - I only knew because I happen to be familiar with youtube-dl.