shivasiddharth/GassistPi #885: Language doesn't change

Oscaretto, 2019-04-29:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000003ab45978
```
```
OS

`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"` 
```

### Describe the bug:
```
I have changed language in config.yaml in "it-IT" but also restarting the Raspberry it speak only in English.
```

[... 16 lines cut ...]
### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
No problem
```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
2019-04-29 21:48:06,198 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token 2019-04-29 21:48:06,213 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443 2019-04-29 21:48:27,142 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token 2019-04-29 21:48:27,154 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443 2019-04-29 21:48:38,873 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token 2019-04-29 21:48:38,885 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443 2019-04-29 21:48:39,631 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None 2019-04-29 21:51:36,467 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443 2019-04-29 21:51:37,000 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None 2019-04-29 21:51:37,579 : DEBUG : https://translate.google.com:443 "GET /translate_a/single?dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&hl=it&ssel=0&otf=1&sl=en&q=Hey%2C+you+need+to+enter+your+google+cloud+api+in+the+config+file+first.&client=t&tsel=0&oe=UTF-8&tk=138627.296312&tl=it&ie=UTF-8 HTTP/1.1" 200 None 2019-04-29 21:51:37,587 : DEBUG : slow: False 2019-04-29 21:51:37,601 : DEBUG : text: Ehi, devi prima inserire la tua API di google cloud nel file di configurazione. 2019-04-29 21:51:37,602 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x70a0f198>, <function end_of_line at 0x70a0f588>, <function abbreviations at 0x70a0f5d0>, <function word_sub at 0x70a0f618>] 2019-04-29 21:51:37,603 : DEBUG : lang_check: True 2019-04-29 21:51:37,603 : DEBUG : lang: it 2019-04-29 21:51:37,604 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\;|\\…|\\(|\\‥|\\。|\\:|\\،|\\，|\\—|\\[|\\\n|\\¡|\\¿|\\：|\\]|\\)|\\、', re.IGNORECASE) from: [<function tone_marks at 0x70a0f6a8>, <function period_comma at 0x70a0f6f0>, <function other_punctuation at 0x70a0f738>]> 2019-04-29 21:51:37,629 : DEBUG : Starting new HTTP connection (1): translate.google.com:80 2019-04-29 21:51:37,860 : DEBUG : http://translate.google.com:80 "GET / HTTP/1.1" 302 226 2019-04-29 21:51:37,883 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443 2019-04-29 21:51:38,500 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None 2019-04-29 21:51:38,761 : DEBUG : Starting new HTTP connection (1): translate.google.com:80 2019-04-29 21:51:39,020 : DEBUG : http://translate.google.com:80 "GET //translate/releases/twsfe_w_20190422_RC01/r/js/translate_m.js HTTP/1.1" 200 198678 2019-04-29 21:51:39,508 : DEBUG : langs: {'fr-fr': 'French (France)', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'sq': 'Albanian', 'te': 'Telugu', 'et': 'Estonian', 'km': 'Khmer', 'ne': 'Nepali', 'en-ie': 'English (Ireland)', 'cy': 'Welsh', 'sk': 'Slovak', 'sv': 'Swedish', 'fi': 'Finnish', 'jw': 'Javanese', 'sr': 'Serbian', 'es-es': 'Spanish (Spain)', 'en-gb': 'English (UK)', 'hi': 'Hindi', 'de': 'German', 'lv': 'Latvian', 'vi': 'Vietnamese', 'pt': 'Portuguese', 'en-us': 'English (US)', 'ko': 'Korean', 'en-tz': 'English (Tanzania)', 'hr': 'Croatian', 'es': 'Spanish', 'en-uk': 'English (UK)', 'fr': 'French', 'su': 'Sundanese', 'en-gh': 'English (Ghana)', 'si': 'Sinhala', 'ru': 'Russian', 'bs': 'Bosnian', 'id': 'Indonesian', 'en-ph': 'English (Philippines)', 'en-nz': 'English (New Zealand)', 'ml': 'Malayalam', 'cs': 'Czech', 'en-za': 'English (South Africa)', 'en-au': 'English (Australia)', 'my': 'Myanmar (Burmese)', 'pt-br': 'Portuguese (Brazil)', 'it': 'Italian', 'la': 'Latin', 'sw': 'Swahili', 'en': 'English', 'ta': 'Tamil', 'en-in': 'English (India)', 'zh-cn': 'Chinese (Mandarin/China)', 'th': 'Thai', 'no': 'Norwegian', 'ro': 'Romanian', 'en-ng': 'English (Nigeria)', 'mk': 'Macedonian', 'hu': 'Hungarian', 'mr': 'Marathi', 'bn': 'Bengali', 'tr': 'Turkish', 'es-us': 'Spanish (United States)', 'el': 'Greek', 'ar': 'Arabic', 'uk': 'Ukrainian', 'hy': 'Armenian', 'en-ca': 'English (Canada)', 'da': 'Danish', 'nl': 'Dutch', 'af': 'Afrikaans', 'ja': 'Japanese', 'ca': 'Catalan', 'is': 'Icelandic', 'pt-pt': 'Portuguese (Portugal)', 'tl': 'Filipino', 'fr-ca': 'French (Canada)', 'pl': 'Polish', 'eo': 'Esperanto'} 2019-04-29 21:51:39,509 : DEBUG : pre-processing: <function tone_marks at 0x70a0f198> 2019-04-29 21:51:39,515 : DEBUG : pre-processing: <function end_of_line at 0x70a0f588> 2019-04-29 21:51:39,516 : DEBUG : pre-processing: <function abbreviations at 0x70a0f5d0> 2019-04-29 21:51:39,521 : DEBUG : pre-processing: <function word_sub at 0x70a0f618> 2019-04-29 21:51:39,521 : DEBUG : text_parts: 1 2019-04-29 21:51:39,530 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443 2019-04-29 21:51:40,138 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None 2019-04-29 21:51:40,324 : DEBUG : payload-0: {'textlen': 79, 'ttsspeed': 1, 'total': 1, 'tk': '880822.780365', 'idx': 0, 'client': 'tw-ob', 'tl': 'it', 'q': 'Ehi, devi prima inserire la tua API di google cloud nel file di configurazione.', 'ie': 'UTF-8'} 2019-04-29 21:51:40,344 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443 2019-04-29 21:51:41,024 : DEBUG : https://translate.google.com:443 "GET /translate_tts?textlen=79&ttsspeed=1&total=1&tk=880822.780365&idx=0&client=tw-ob&tl=it&q=Ehi%2C+devi+prima+inserire+la+tua+API+di+google+cloud+nel+file+di+configurazione.&ie=UTF-8 HTTP/1.1" 200 26016 2019-04-29 21:51:41,089 : DEBUG : headers-0: {'Accept-Encoding': 'gzip, deflate', 'Connection': 'keep-alive', 'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47. [... cut ...] en 2019-04-29 22:12:08,335 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443 2019-04-29 22:12:09,228 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None 2019-04-29 22:15:51,047 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token 2019-04-29 22:15:51,059 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443 2019-04-29 22:15:51,604 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

shivasiddharth, 2019-04-30:
@aandroide May be you can help him ?

Flog786, 2019-05-22:
Hello, same problem here after modification on language in the config.yaml and restart of the PI the general language doesn't change to FR-fr

sierdw84, 2019-05-29:
Already change in this?

shivasiddharth, 2019-05-31:
Language change in the config.yaml file is only for the speech to text that is used. To change the assistant's voice, you must do it in the Home app on your phone.

aandroide, 2019-06-07:
> @aandroide May be you can help him ?

Sorry I'm late, but I'm messed up with the job, I can't emulate the problem. to me it works properly. lately there are several updates on google servers.
the problem has been solved?

shivasiddharth, 2020-03-07:
Closing this for inactivity

closed 2020-03-07 by shivasiddharth