shivasiddharth/GassistPi #890: Youtube taking so long to load then failed

BarryAlvord, 2019-05-01:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a32082
Serial		: 0000000018a688d3

```

Paste your Board and OS details here.   
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
Youtube taking so long to load the song then failed
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
Yes
### 2. Is the issue related to assistant installation ?
Yes/No
No
### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
Yes
### 4. Is the assistant service restarting automatically?
Yes/No
No
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}

ON_NO_RESPONSE

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "idiot"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_RENDER_RESPONSE:
  {"text": "I'm really sorry 😞", "type": 0}

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Hong Kong one from YouTube"}

stopping vlc

#########
And it stucks right there for 10 min then the assistant said something about lost connection
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-05-01:
Attach the log file

BarryAlvord, 2019-05-01:
@shivasiddharth i just Attach the log file recently please help me

shivasiddharth, 2019-05-01:
Your youtube_dl must be outdated. Check https://github.com/shivasiddharth/GassistPi/issues/521#issuecomment-420043868

BarryAlvord, 2019-05-02:
After I upgraded it still run into the same issue like before

shivasiddharth, 2019-05-02:
Post your new logs

BarryAlvord, 2019-05-02:
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3138702/GassistPi.log)
Here is my full log after i delete the old log and have the new log thet have the error 
pls hlep me

shivasiddharth, 2019-05-02:
Share the output of 
```
source env/bin/activate
pip freeze
```

BarryAlvord, 2019-05-02:
(env) root@raspberrypi:/home/pi# pip freeze
adafruit-io==2.1
aftership==0.2
appdirs==1.4.3
asn1crypto==0.24.0
beautifulsoup4==4.7.1
bs4==0.0.1
cachetools==3.1.0
casttube==0.2.0
certifi==2019.3.9
cffi==1.12.3
chardet==3.0.4
click==6.7
colorzero==1.1
cryptography==2.4.2
decorator==4.4.0
enum34==1.1.6
feedparser==5.2.1
future==0.17.1
futures==3.1.1
gmusicapi==12.1.0
google-api-core==1.10.0
google-api-python-client==1.7.8
google-assistant-grpc==0.2.1
google-assistant-library==1.0.1
google-assistant-sdk==0.5.1
google-auth==1.6.3
google-auth-httplib2==0.0.3
google-auth-oauthlib==0.3.0
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.5.10
googletrans==2.3.0
gpiozero==1.5.0
gpsoauth==0.4.1
grpcio==1.20.1
gTTS==2.0.2
gTTS-token==1.1.3
httplib2==0.12.3
idna==2.8
ifaddr==0.1.6
kodi-json==1.0.0
lxml==4.3.3
MechanicalSoup==0.11.0
mock==2.0.0
mutagen==1.42.0
numpy==1.16.3
oauth2client==4.1.3
oauthlib==3.0.1
pafy==0.5.4
paho-mqtt==1.4.0
pathlib2==2.3.3
pbr==5.2.0
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.7.1
psutil==5.6.2
pushbullet.py==0.11.0
pyasn1==0.4.5
pyasn1-modules==0.2.5
PyAudio==0.2.11
PyChromecast==3.2.1
pycparser==2.19
pycryptodomex==3.7.2
python-dateutil==2.8.0
python-magic==0.4.15
python-vlc==3.0.6109
pytz==2019.1
PyYAML==3.13
requests==2.21.0
requests-oauthlib==1.2.0
RPi.GPIO==0.6.3
rsa==4.0
six==1.12.0
sounddevice==0.3.13
soupsieve==1.9.1
spidev==3.4
spotipy==2.4.4
tenacity==4.12.0
uritemplate==3.0.0
urllib3==1.24.2
validictory==1.1.2
websocket-client==0.56.0
youtube-dl==2019.4.30
zeroconf==0.22.0

@shivasiddharth here is the output

shivasiddharth, 2019-05-02:
Can you manually start the assistant, give the youtube command and then post here the contents of your terminal ?

BarryAlvord, 2019-05-02:
Here is my terminal output:
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Hong Kong one by Plan B from YouTube"}

stopping vlc
#It start freezing right here for so long (for 10 min)

High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
hong kong one by plan b
https://r3---sn-8qj-i5ok.googlevideo.com/videoplayback?id=o-AEkFcWxdWXeNLn8blYZ6zvZkZd1y-Kt54-kG7cBb6TvI&itag=251&source=youtube&requiressl=yes&mm=31%2C26&mn=sn-8qj-i5ok%2Csn-i3b7knld&ms=au%2Conr&mv=m&pl=20&ei=Zx7LXMzeONrGgAPhx6q4Bg&initcwndbps=861250&mime=audio%2Fwebm&gir=yes&clen=4507582&dur=264.821&lmt=1539376375051084&mt=1556815379&fvip=3&keepalive=yes&c=WEB&txp=5511222&ip=14.177.11.222&ipbits=0&expire=1556837064&sparams=ip%2Cipbits%2Cexpire%2Cid%2Citag%2Csource%2Crequiressl%2Cmm%2Cmn%2Cms%2Cmv%2Cpl%2Cei%2Cinitcwndbps%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&signature=3E8A33040026FDB842B8FC68ABE0086A8F862E04.BFBC22B622C3D88C468FEE45A12CB8FADDFDE37C&key=yt8&ratebypass=yes
[01d43880] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

[6874d6d0] main tls client error: TLS session handshake timeout
[6874d6d0] main tls client error: connection error: Resource temporarily unavailable

BarryAlvord, 2019-05-02:
Here is another terminal output:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Heaven by Kane Brown on YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
heaven by kane brown on youtub
timed out

shivasiddharth, 2020-11-02:
Closing this for inactivity.

closed 2020-11-02 by shivasiddharth