shivasiddharth/GassistPi #891: Segmentation fault when kodi Enabled in config.yaml

tdvac, 2019-05-03:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  

Paste your Board and OS details here.   
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000f35135d5

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
Hello everyone.
With the GassistPi version of February 3, when KODI is activated in config.yaml, I get a segmentation error when returning to listen.
Otherwise it worked fine with the earlier version of GassistPi with the same config.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "quelle heure est-il"}

Kodi TV box not online
ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Fatal Python error: Segmentation fault

Thread 0x63170470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x62970470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76f70010 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Segmentation fault

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
2019-05-03 10:18:04,250 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-05-03 10:18:04,279 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-05-03 10:18:04,462 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-05-03 10:18:17,768 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
2019-05-03 10:18:20,936 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200

shivasiddharth, 2019-05-03:
> Kodi TV box not online

Clearly it says here that your Kodi is not online. Sorry I still dont understand your issue.

tdvac, 2019-05-03:
KODI is managed by another RPI3B + (192.168.0.43:8200).
With the previous version of GassistPi, even if Kodi was not online I had no crash and Google returned to listening.
With a new installation integrating the latest version of GassistPi, there is systematic crash that KODI is online or not when it is enabled in config.yaml.

shivasiddharth, 2019-05-03:
> Kodi TV box not online
> ON_END_OF_UTTERANCE
> 
> ON_END_OF_UTTERANCE
> 
> ON_RECOGNIZING_SPEECH_FINISHED:
>   {"text": "quelle heure est-il"}
> 
> Kodi TV box not online
> ON_RESPONDING_STARTED:
>   {"is_error_response": false}
> 
> ON_RESPONDING_FINISHED
> 
> ON_CONVERSATION_TURN_FINISHED:
>   {"with_follow_on_turn": false}
> 

Assistant is of course returning back. Attach the log files let me check as to what else is happening.

tdvac, 2019-05-03:
2019-05-03 14:22:09,484 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-05-03 14:22:09,501 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-05-03 14:22:09,679 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-05-03 14:22:20,186 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
2019-05-03 14:22:23,465 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
2019-05-03 14:22:42,690 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
2019-05-03 14:22:46,024 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
2019-05-03 14:22:49,059 : DEBUG : Starting new HTTP connection (1): 192.168.0.43:8200
____________________________________________________________________________________________________________
Apparently this problem only occurs during the first installation (KODI stopped on the second RPI).
It is imperative that the link is recognized a first time between GassistPi and KODi, so that the error is no longer involved.
After several tests (KODI stopped on the second RPI), the segmentation error no longer appears.
To confirm this problem, I realized three other new installations which required the same manipulations so that the error disappears.
Would there be a flag that would mark the presence of KODI during the first connection?

shivasiddharth, 2019-05-17:
I had programmed the assistant to check for Kodi's presence each and every time a command is sent. That is why you are seeing the Kodi not online command. after each conversation. Still i am unable to understand the problem, I am getting lost here. 

1. You start the assistant with Kodi online everything is working.
2. Now Kodi goes offline and you get segmentation fault.
3. You reboot the assistant with Kodi offline and now you dont get the segmentation fault.

Is my understanding correct?

tdvac, 2019-05-18:
Hello everyone,
In fact, I also tried to redo a GassistPi installation without KODI and disable it in config.yaml. it works perfectly.
Otherwise if I enable KODI in config.yaml, I get the segmentation error.
The absence of KODI is well recognized because the message "kodi not online" is displayed.
Then I start KODi on the second RPI, I restart GassistPi who recognizes KODI and works.
Now, if I turn off the second RPI and restart GassistPi, I do not have a segmentation error as before and it works perfectly and at each startup.
KODI must be recognized a first time and then the error does not appear when it is offline.
Sorry i'm french and i write you by google translation

shivasiddharth, 2020-03-07:
Closing this for inactivity.

closed 2020-03-07 by shivasiddharth

gowdhamGM, 2020-03-07:
Thank you for your response