shivasiddharth/GassistPi #893: After questioning the response gets cut off and freezes

NCPlyn, 2019-05-04:
### Board and OS details:
```
cat /proc/cpuinfo:
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 000000002936ea84
---------------------------------------
cat /etc/os-release:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
After manually launching the GassistPi, got two PulseAudio errors.
Then, when I say something, I hear something like speaking for about 0.5 sec and then nothing. CTRL+C doesn't work, I have to close and open new ssh conection.
Using RPi Zero W with latest raspbian-stretch-lite on 16GB sdcard.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
Something between :)

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/ghome/GassistPi/src/pushbutton.py --project-id XXXXXX  --device-model-id XXXXXX
```

### Paste the contents of the terminal below
```
pi@raspberrypi:~/ghome/GassistPi/src $ /home/${USER}/env/bin/python -u /home/${USER}/ghome/GassistPi/src/pushbutton.py --project-id XXXX  --device-model-id XXXXX
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-05-04 18:21:25,762 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[02117bb0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[02121310] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 1 detected at time: 2019-05-04 18:21:54
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
what\'s the time
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

NCPlyn, 2019-05-04:
OK, I'm dumb...

`sudo apt-get install pulseaudio` solved it....

but still i get this: 

```
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2019-05-04 19:03:36,259 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
```
Any help for this?

NCPlyn, 2019-05-05:
Also, is there way to get output on GPIO when the mic is muted? And is there way to have button for trigger while have snowboy activated? Thanks for reply.

bachoo786, 2019-06-12:
@NCPlyn  did you manage to resolve this? I have the same issue as you