shivasiddharth/GassistPi #898: Installation failure

johnryanlee, 2019-05-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 100.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 100.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 100.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 100.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Allwinner sun8i Family
Revision        : 0000
Serial          : 02c00042f1153402

PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

### Describe the bug:
```
Crashes on install at lxml
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-QMv3zq/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -I/usr/include/libxml2 -Isrc -Isrc/lxml/includes -I/home/ryanlee/env/include -I/usr/include/python3.5m -c src/lxml/etree.c -o build/temp.linux-armv7l-3.5/src/lxml/etree.o -w
  arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1)
  Please submit a full bug report,
  with preprocessed source if appropriate.
  See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
  Compile failed: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
  creating tmp
  cc -I/usr/include/libxml2 -I/usr/include/libxml2 -c /tmp/xmlXPathInitjzc1s4ph.c -o tmp/xmlXPathInitjzc1s4ph.o
  /tmp/xmlXPathInitjzc1s4ph.c:2:1: warning: return type defaults to 'int' [-Wimplicit-int]
   main (int argc, char **argv) {
   ^~~~
  cc tmp/xmlXPathInitjzc1s4ph.o -lxml2 -o a.out
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
  ----------------------------------------
  ERROR: Failed building wheel for lxml

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory

shivasiddharth, 2019-05-08:
This looks like an Orange Pi board. Which OS are you using?

johnryanlee, 2019-05-08:
Armbian Stretch
I have monitored the RAM during the process and it runs out from what I can see, so it looks like a resource issue.  The board has 512MB RAM
I've installed Armbian Bionic now to see if there's any difference between the OS's

jabulonboaz, 2019-05-15:
i have the same issue buildng on a raspberry pi a+ it als has 512MB of memory, what shuld i do please?

johnryanlee, 2019-05-15:
You need to increase your swap storage because I’m not aware of any means to increase the physical RAM.

What distro are you using?

jabulonboaz, 2019-05-15:
I am using raspbian stretch OS latest version just started building it here is my log, how do icrease the swap storage on the Pi a+? 

Building wheel for lxml (setup.py) ... error
  ERROR: Complete output from command /home/pi/env/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-uoxvnotw/lxml/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-dz7dje_1 --python-tag cp35:
  ERROR: Building lxml version 4.3.3.
  Building without Cython.
  Using build configuration of libxslt 1.1.29
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/sax.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/builder.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/cssselect.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/__init__.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-3.5/lxml
  creating build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-3.5/lxml/includes
  creating build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/clean.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/builder.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/diff.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/defs.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/html5parser.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv7l-3.5/lxml/html
  copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-3.5/lxml/html
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron
  copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv7l-3.5/lxml/isoschematron
  copying src/lxml/etree.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/etree_api.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/lxml.etree.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv7l-3.5/lxml
  copying src/lxml/includes/xmlschema.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/etreepublic.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/htmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/config.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/tree.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/xmlerror.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/uri.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/lxml-version.h -> build/lib.linux-armv7l-3.5/lxml/includes
  copying src/lxml/includes/etree_defs.h -> build/lib.linux-armv7l-3.5/lxml/includes
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
  copying src/lxml/isoschematron/resources/rng/iso-schematron.rng -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  copying src/lxml/isoschematron/resources/xsl/RNG2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  copying src/lxml/isoschematron/resources/xsl/XSD2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
  creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
  copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_abstract_expand.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
[... 100 lines cut ...]
[... cut ...]
ERROR: Command "/home/pi/env/bin/python -u -c 'import setuptools, tokenize;__file__='"'"'/tmp/pip-install-uoxvnotw/lxml/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-r34w_9s9/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/env/include/site/python3.5/lxml" failed with error code 1 in /tmp/pip-install-uoxvnotw/lxml/
pi@raspberrypi:~ $

jabulonboaz, 2019-05-15:
It fixed it for me too thanks for this! i increased the swap file size from 100 to 500MB. i am using Rspian strtch OS.

jbrepogmailcom, 2019-07-20:
Dear jabulonboaz may I please ask you how you increased swap on Orange Pi? Thank you very much, Jan

bachoo786, 2019-07-21:
@johnryanlee @jabulonboaz how did you guys increase the swap file? I guess me and @jbrepogmailcom are facing issues and we need to increase the swap file but dont know how to.

Any help will be appreciated. Thanks.

jbrepogmailcom, 2019-07-21:
Hello I used something like this:

sudo dd if=/dev/zero of=/var/swap bs=1M count=1024
sudo chmod 0600 /var/swap
sudo mkswap /var/swap
sudo swapon /var/swap

bachoo786, 2019-07-21:
@jbrepogmailcom thanks for this it worked for me now.

May I ask you..can you connect to wifi using your orange pi zero? I presume you are on armbian stretch with kernel 4.14?

jbrepogmailcom, 2019-07-21:
That was first thing aftet flashing new os. Connect over ethermet and install:
sudo apt-get install wicd wicd-curses
Then run sudo wicd-curses
And configure wifi. Be sure you can connect over wifi after boot before you cutoff ethernet :-)

shivasiddharth, 2020-11-02:
Closing this for inactivity.

closed 2020-11-02 by shivasiddharth