shivasiddharth/GassistPi #907: Assistant stops after detecting OK Google hotword

vishalds2003, 2019-05-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo 
cat /etc/os-release  
```
```
Board

processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a21041
Serial		: 00000000df98f327
   
OS DETAILS

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:
```
The assistant responds when triggered with the OK Google hotword. Then, when we say some query, example, what's the time now, it automatically crashes. Seems like it's a problem with snowboy, is it?
When started using systemctl during boot, it automatically restarts after crashing when we give some query.  
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Answer is NO
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes when started using systemctl

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's time now"}

Fatal Python error: Segmentation fault

Thread 0x669ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x659ff470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76f00010 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1019 in main
  File "/home/pi/GassistPi/src/main.py", line 1027 in <module>
Segmentation fault

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id <device model id>

Also, it crashes when started on boot using systemctl too
```

### Paste the contents of the terminal below
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's time now"}

Fatal Python error: Segmentation fault

Thread 0x669ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x659ff470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76f00010 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1019 in main
  File "/home/pi/GassistPi/src/main.py", line 1027 in <module>
Segmentation fault

```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3224158/GassistPi.log)

shivasiddharth, 2019-05-31:
Looks like you have exceeded the free limit.

vishalds2003, 2019-06-01:
What's the free limit..?
Is there a limit for using the assistant?

vishalds2003, 2019-06-01:
Are you talking about the Google API request limit ...?

shivasiddharth, 2019-06-05:
Assistant API limit is 500 calls for 24hrs.. If you have exceeded that the assistant will not work. Or else check your keywords file if you have something very basic as your keyword.

vishalds2003, 2019-06-05:
But it plays the response tone when I say Ok Google ....
But it just doesn't listen for user query...? Is that a symptom of exceeding the limit ? Also, if it is 500 calls for 24 hrs, then, won't it recharge with the next 24 hours...?

shivasiddharth, 2019-06-05:
It will be reset every 24 hours. If you have made any changes to the scripts undo them and check again.

vishalds2003, 2019-06-05:
Ok

vishalds2003, 2019-06-07:
Anyway, I now began a new project and reinstalled the assistant. So, now it's working. But I don't know what caused the issue. Will either reopen this issue or open another issue if this occurs again....
Or, I will tell at Gitter.

closed 2019-06-07 by vishalds2003