shivasiddharth/GassistPi #909: Assistant Crashed

Friday8229, 2019-05-30:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   

pi@raspberrypi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 89.60
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000054341172

pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[... 51 lines cut ...]
ain.py --device_model_id 'zero2500-zero250-w48xlq'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'data': {'entries': [{'value': 'false', 'kind': 'sj#configEntry', 'key': 'canManageFP'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowLocalSuggestions'}, {'value': '3', 'kind': 'sj#configEntry', 'key': 'purchaseAccess'}, {'value': '30000', 'kind': 'sj#configEntry', 'key': 'wsVdtMillis'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'sideloadedPlaybackLogging'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isWarmWelcomeDisabledForWoodstock'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'canCancelSubscription'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'nashvillebg'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isVsAvailable'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'canBrowseSt'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'appstoreRatingPrompt'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellDrawerCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDrawerDownloadMessage'}, {'value': '5000', 'kind': 'sj#configEntry', 'key': 'wsVatMillis'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'nuqRequiredForWsSignup'}, {'value': 'GET MORE', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellNewReleasesCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellListenNowCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'enableWsExtras'}, {'value': 'Subscribe to get millions of songs on-demand, offline, and ad-free', 'kind': 'sj#configEntry', 'key': 'upsellSignedOutMessage'}, {'value': '500', 'kind': 'sj#configEntry', 'key': 'topListMaxSize'}, {'value': '2 skips left', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showLMark'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'conciergeListenNow'}, {'value': '6', 'kind': 'sj#configEntry', 'key': 'minimumAppStartCountForFreeUpsell'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellPreviousCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDownloadMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showYoutubeRedInfoCard'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'concListenNow'}, {'value': '300', 'kind': 'sj#configEntry', 'key': 'instantSearchDelay'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellChartsMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isPodcastSyncEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellChartsCtaMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showViOSL'}, {'value': '604800', 'kind': 'sj#configEntry', 'key': 'topListUpdatePeriodSeconds'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isSuggestDeletionEnabled'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowServerSuggestions'}, {'value': '30', 'kind': 'sj#configEntry', 'key': 'situationsResponseTTLMinutes'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isFoplessNautilusUser'}, {'value': '1459804296734', 'kind': 'sj#configEntry', 'key': 'appStartCountForFreeUpsellRevisionNumber'}, {'value': 'Choose your language preference here, too.', 'kind': 'sj#configEntry', 'key': 'mgRPBDesc'}, {'value': '1', 'kind': 'sj#configEntry', 'key': 'partridgeManagement'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showB'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showA'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldEnableThriller'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpMessage'}, {'value': '100', 'kind': 'sj#configEntry', 'key': 'topListPageSize'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isHallmarkEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpCtaMessage'}, {'value': '2', 'kind': 'sj#configEntry', 'key': 'uploadAccess'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isNautilusAvailable'}, {'value': '2', 'kind': 'sj#configEntry', 'key': 'nautilusAccess'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isQuizEnabledForWoodstockOnSignup'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showO'}, {'value': '200000', 'kind': 'sj#configEntry', 'key': 'disabledConnectivityTimeoutMs'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'hasPControlSettings'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'wasRecentlyInFoplessNautilus'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showV'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellSignedOutCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'forceSignupFlow'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isNautilusUser'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isLNEnabled'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'wasEverInNautilus'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'ytContent'}, {'value': 'GET MORE', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitOneLeftCtaMessage'}, {'value': '5000', 'kind': 'sj#configEntry', 'key': 'wsAatMillis'}, {'value': '1', 'kind': 'sj#configEntry', 'key': 'partridgeAccess'}, {'value': 'LISTEN_NOW', 'kind': 'sj#configEntry', 'key': 'warmWelcomeButtonText'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellPrerollVideoCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isTrAvailable'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isInAppMessageEnabled'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'disableInfoCardsOnDayOne'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'canRC'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': ' [... cut ...]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 28, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
No log file found in the tmp directory

shivasiddharth, 2019-06-05:
Duplicate of #839

Friday8229, 2019-06-07:
I saw the issue #866 and i commented the lines 27 and 28 in the GassistPi/src/mediaplayer.py and my assistant started working normally but when I asked to play a music from youtube there is no response after it says getting youtube link. The problem starts now my assistant was working fine except youtube and as you mentioned in issue #890 I updated the youtube.dll and my assistant and after rebooting t started acting strangely when I say "ok google" it took about 5 sec to hear the ding tone and it is very slow now when i say to play a song from youtube at first it says "something went wrong" and then it says getting youtube link and then nothing happens . Thanks in advance

Friday8229, 2019-06-07:
HI shivasiddharth , my assistant started working normally after rebooting it 3 times but the youtube issue is still the same. It works normally except youtube and I want to know that can I increase the speed of my assistant.

aandroide, 2019-06-07:
check if your youtube apikey is still active from the google console, after a few months of non-use google deactivates it and could cause the problem you are experiencing.

Friday8229, 2019-06-09:
Hi, aandroide my YouTube apikey is still active from the Google console.

shivasiddharth, 2019-06-11:
The assistant will not reboot for youtube issues, it will say "Encountered an exception please check the logs". But it will not restart. Check if you have enabled youtube data api.

Friday8229, 2019-06-11:
Sorry but there is a miss understanding I was saying that it started working correctly after I rebooted it three times manuallay. My youtube issue is still the same I have checked the youtube api it is enabled. 

thanks in advance

shivasiddharth, 2019-06-12:
I need to see the terminal output for youtube issue. Can you give the youtube streaming command and then share the terminal output?

Friday8229, 2019-06-12:
This is the terminal output 

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'zero2500-zero250-w48xlq'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#configList', 'data': {'entries': [{'value': 'false', 'kind': 'sj#configEntry', 'key': 'canManageFP'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowLocalSuggestions'}, {'value': '3', 'kind': 'sj#configEntry', 'key': 'purchaseAccess'}, {'value': '30000', 'kind': 'sj#configEntry', 'key': 'wsVdtMillis'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'sideloadedPlaybackLogging'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isWarmWelcomeDisabledForWoodstock'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'canCancelSubscription'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'nashvillebg'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isVsAvailable'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'canBrowseSt'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'appstoreRatingPrompt'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellDrawerCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDrawerDownloadMessage'}, {'value': '5000', 'kind': 'sj#configEntry', 'key': 'wsVatMillis'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'nuqRequiredForWsSignup'}, {'value': 'GET MORE', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellNewReleasesCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellListenNowCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'enableWsExtras'}, {'value': 'Subscribe to get millions of songs on-demand, offline, and ad-free', 'kind': 'sj#configEntry', 'key': 'upsellSignedOutMessage'}, {'value': '500', 'kind': 'sj#configEntry', 'key': 'topListMaxSize'}, {'value': '2 skips left', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showLMark'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'conciergeListenNow'}, {'value': '6', 'kind': 'sj#configEntry', 'key': 'minimumAppStartCountForFreeUpsell'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellPreviousCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDownloadMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showYoutubeRedInfoCard'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'concListenNow'}, {'value': '300', 'kind': 'sj#configEntry', 'key': 'instantSearchDelay'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellChartsMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isPodcastSyncEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellChartsCtaMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showViOSL'}, {'value': '604800', 'kind': 'sj#configEntry', 'key': 'topListUpdatePeriodSeconds'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isSuggestDeletionEnabled'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowServerSuggestions'}, {'value': '30', 'kind': 'sj#configEntry', 'key': 'situationsResponseTTLMinutes'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isFoplessNautilusUser'}, {'value': '1459804296734', 'kind': 'sj#configEntry', 'key': 'appStartCountForFreeUpsellRevisionNumber'}, {'value': 'Choose your language preference here, too.', 'kind': 'sj#configEntry', 'key': 'mgRPBDesc'}, {'value': '1', 'kind': 'sj#configEntry', 'key': 'partridgeManagement'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showB'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showA'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldEnableThriller'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpMessage'}, {'value': '100', 'kind': 'sj#configEntry', 'key': 'topListPageSize'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isHallmarkEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpCtaMessage'}, {'value': '2', 'kind': 'sj#configEntry', 'key': 'uploadAccess'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isNautilusAvailable'}, {'value': '2', 'kind': 'sj#configEntry', 'key': 'nautilusAccess'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isQuizEnabledForWoodstockOnSignup'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showO'}, {'value': '200000', 'kind': 'sj#configEntry', 'key': 'disabledConnectivityTimeoutMs'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'hasPControlSettings'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'wasRecentlyInFoplessNautilus'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showV'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellSi [... cut ...] dur=216.901&lmt=1555572226846951&mt=1560335589&fvip=1&keepalive=yes&c=WEB&txp=5531432&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cgcr%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=mm%2Cmn%2Cms%2Cmv%2Cnh%2Cpl%2Cinitcwndbps&lsig=AHylml4wRQIhAOhIDVYGQWtxTvgnH8w5ApF4cKKTm4Yv-EB_VO5K8acvAiBFQlTFxVHuVlAzrvYxer3Kv9bc_y7lpcSGejb3lety7w%3D%3D&sig=ALgxI2wwRQIhAMmfb022Nop9KSZpYK4gP_ReekuhgjAzIta-C9DxCHQIAiBww3nQrD0UzZgk_K1WAKNuaDgpzfA_uie2XWFznCamug==&ratebypass=yes'. Check the log for details.

Friday8229, 2019-06-12:
This is my log file

2019-05-26 08:52:29,536 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-05-26 08:52:29,555 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-05-26 08:52:30,281 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-05-26 08:55:31,969 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2019-05-26 08:55:31,998 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2019-05-26 08:55:32,578 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2019-05-26 08:55:48,140 : INFO : Keyword 3 detected at time: 2019-05-26 08:55:48
2019-05-26 08:55:48,256 : INFO : Keyword 3 detected at time: 2019-05-26 08:55:48
2019-05-26 08:55:50,205 : DEBUG : lang: en
2019-05-26 08:55:50,208 : DEBUG : lang_check: True
2019-05-26 08:55:50,209 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\：|\\;|\\¡|\\،|\\:|\\—|\\]|\\、|\\(|\\\n|\\‥|\\¿|\\。|\\…|\\[|\\，|\\)', re.IGNORECASE) from: [<function tone_marks at 0x70818030>, <function period_comma at 0x70818078>, <function other_punctuation at 0x708180c0>]>
2019-05-26 08:55:50,210 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x7080dae0>, <function end_of_line at 0x7080ded0>, <function abbreviations at 0x7080df18>, <function word_sub at 0x7080df60>]
2019-05-26 08:55:50,211 : DEBUG : slow: False
2019-05-26 08:55:50,213 : DEBUG : text: Getting youtube link
2019-05-26 08:55:50,253 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2019-05-26 08:55:50,517 : DEBUG : http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2019-05-26 08:55:50,556 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2019-05-26 08:55:51,237 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2019-05-26 08:55:51,710 : DEBUG : Starting new HTTP connection (1): translate.google.com:80
2019-05-26 08:55:51,909 : DEBUG : http://translate.google.com:80 "GET //translate/releases/twsfe_w_20190603_RC02/r/js/translate_m.js HTTP/1.1" 200 199300
2019-05-26 08:55:52,688 : DEBUG : langs: {'cs': 'Czech', 'sq': 'Albanian', 'da': 'Danish', 'mk': 'Macedonian', 'fi': 'Finnish', 'th': 'Thai', 'ja': 'Japanese', 'pl': 'Polish', 'fr': 'French', 'eo': 'Esperanto', 'te': 'Telugu', 'hy': 'Armenian', 'ta': 'Tamil', 'km': 'Khmer', 'tl': 'Filipino', 'sv': 'Swedish', 'el': 'Greek', 'sr': 'Serbian', 'my': 'Myanmar (Burmese)', 'en-in': 'English (India)', 'si': 'Sinhala', 'ko': 'Korean', 'en-ca': 'English (Canada)', 'en-ph': 'English (Philippines)', 'uk': 'Ukrainian', 'nl': 'Dutch', 'pt-br': 'Portuguese (Brazil)', 'bs': 'Bosnian', 'pt': 'Portuguese', 'en-gb': 'English (UK)', 'it': 'Italian', 'en-ie': 'English (Ireland)', 'de': 'German', 'is': 'Icelandic', 'en-ng': 'English (Nigeria)', 'fr-ca': 'French (Canada)', 'hi': 'Hindi', 'ru': 'Russian', 'vi': 'Vietnamese', 'ca': 'Catalan', 'ar': 'Arabic', 'la': 'Latin', 'en-uk': 'English (UK)', 'en-gh': 'English (Ghana)', 'ml': 'Malayalam', 'id': 'Indonesian', 'cy': 'Welsh', 'en-nz': 'English (New Zealand)', 'en-tz': 'English (Tanzania)', 'fr-fr': 'French (France)', 'bn': 'Bengali', 'es': 'Spanish', 'es-es': 'Spanish (Spain)', 'sk': 'Slovak', 'en-us': 'English (US)', 'ro': 'Romanian', 'sw': 'Swahili', 'en-za': 'English (South Africa)', 'mr': 'Marathi', 'tr': 'Turkish', 'pt-pt': 'Portuguese (Portugal)', 'et': 'Estonian', 'no': 'Norwegian', 'en-au': 'English (Australia)', 'su': 'Sundanese', 'en': 'English', 'ne': 'Nepali', 'lv': 'Latvian', 'zh-cn': 'Chinese (Mandarin/China)', 'es-us': 'Spanish (United States)', 'jw': 'Javanese', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'af': 'Afrikaans', 'hu': 'Hungarian', 'hr': 'Croatian'}
2019-05-26 08:55:52,690 : DEBUG : pre-processing: <function tone_marks at 0x7080dae0>
2019-05-26 08:55:52,701 : DEBUG : pre-processing: <function end_of_line at 0x7080ded0>
2019-05-26 08:55:52,702 : DEBUG : pre-processing: <function abbreviations at 0x7080df18>
2019-05-26 08:55:52,717 : DEBUG : pre-processing: <function word_sub at 0x7080df60>
2019-05-26 08:55:52,722 : DEBUG : text_parts: 1
2019-05-26 08:55:52,757 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2019-05-26 08:55:53,424 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2019-05-26 08:55:53,633 : DEBUG : payload-0: {'client': 'tw-ob', 'ttsspeed': 1, 'q': 'Getting youtube link', 'textlen': 20, 'ie': 'UTF-8', 'idx': 0, 'total': 1, 'tk': '766531.864081', 'tl': 'en'}
2019-05-26 08:55:53,679 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2019-05-26 08:55:53,855 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x707fb9c0>, <function end_of_line at 0x707fbdb0>, <function abbreviations at 0x707fbdf8>, <function word_sub at 0x707fbe40>]
2019-05-26 08:55:53,856 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\。|\\、|\\[|\\…|\\，|\\¿|\\：|\\]|\\;|\\\n|\\—|\\‥|\\¡|\\)|\\،|\\(|\\:', re.IGNORECASE) from: [<function tone_marks at 0x707fbed0>, <function period_comma at 0x707fbf18>, <function other_punctuation at 0x707fbf60>]>
2019-05-26 08:55:53,858 : DEBUG : text: Getting youtube link
2019-05-26 08:55:53,859 : DEBUG : slow: False
2019-05-26 08:55:53,860 : DEBUG : lang: en
2019-05-26 08:55:53,861 : DEBUG : lang_check: True
[... cut ...]
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
2019-05-26 08:55:59,530 : INFO : URL being requested: GET https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyBDF63wzB1fcjgJ72GxubQUhNEQsS3YIM0
2019-05-26 08:56:00,651 : INFO : URL being requested: GET https://www.googleapis.com/youtube/v3/search?key=AIzaSyBDF63wzB1fcjgJ72GxubQUhNEQsS3YIM0&q=on+my+way&alt=json&part=id%2Csnippet

Friday8229, 2019-06-13:
hi I am back with a fresh new error now my assistant is not working. The terminal output is

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'zero2500-zero250-w48xlq'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/utils/utils.py", line 655, in __get__
    value, last_update = inst._cache[self.__name__]
KeyError: 'is_subscribed'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/contrib/pyopenssl.py", line 472, in wrap_socket
    cnx.do_handshake()
  File "/home/pi/env/lib/python3.5/site-packages/OpenSSL/SSL.py", line 1915, in do_handshake
    self._raise_ssl_error(self._ssl, result)
  File "/home/pi/env/lib/python3.5/site-packages/OpenSSL/SSL.py", line 1647, in _raise_ssl_error
    _raise_current_error()
  File "/home/pi/env/lib/python3.5/site-packages/OpenSSL/_util.py", line 54, in exception_from_error_queue
    raise exception_type(errors)
OpenSSL.SSL.Error: [('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 603, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 344, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 843, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 370, in connect
    ssl_context=context)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/ssl_.py", line 355, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/contrib/pyopenssl.py", line 478, in wrap_socket
    raise ssl.SSLError('bad handshake: %r' % e)
ssl.SSLError: ("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 641, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 399, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='mclients.googleapis.com', port=443): Max retries exceeded with url: /sj/v2.5/config?hl=en_US&tier=fr&dv=0 (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 20, in <module>
    logged_in=api.oauth_login(Mobileclient.FROM_MAC_ADDRESS)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 183, in oauth_login
    return self._login(session_login, device_id, locale)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 148, in _login
    if self.is_subscribed:
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/utils/utils.py", line 660, in __get__
    value = self.fget(inst)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 111, in is_subscribed
    res = self._make_call(mobileclient.Config)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/mobileclient.py", line 48, in _make_call
    return super(Mobileclient, self)._make_call(protocol, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/clients/shared.py", line 90, in _make_call
    return protocol.perform(self.session, self.validate, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/protocol/shared.py", line 213, in perform
    response = session.send(req_kwargs, required_auth)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/session.py", line 118, in send
    res = self._send_with_auth(req_kwargs, desired_auth, rsession)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/session.py", line 316, in _send_with_auth
    return super(Mobileclient, self)._send_with_auth(req_kwargs, desired_auth, rsession)
  File "/home/pi/env/lib/python3.5/site-packages/gmusicapi/session.py", line 241, in _send_with_auth
[... cut ...]
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='mclients.googleapis.com', port=443): Max retries exceeded with url: /sj/v2.5/config?hl=en_US&tier=fr&dv=0 (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

Friday8229, 2019-06-13:
The log file is not there.

shivasiddharth, 2019-06-14:
This is a network issue. Are you behind a firewall, VPN or proxy ?

Friday8229, 2019-06-14:
I am unable to understand what you are saying. I am using my broadband connection (wifi router ). it was working earlier but it is not working now and can you tell me the way by which I can check that.

shivasiddharth, 2019-06-14:
Open a terminal and type
`curl -v https://www.googleapis.com`
post what you are seeing.

Friday8229, 2019-06-14:
This is the terminal output. 

pi@raspberrypi:~ $ curl -v https://www.googleapis.com
* Rebuilt URL to: https://www.googleapis.com/
*   Trying 172.217.27.202...
* TCP_NODELAY set
* Connected to www.googleapis.com (172.217.27.202) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-ECDSA-AES128-GCM-SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=Mountain View; O=Google LLC; CN=*.googleapis.com
*  start date: May 21 20:50:16 2019 GMT
*  expire date: Aug 13 20:32:00 2019 GMT
*  subjectAltName: host "www.googleapis.com" matched cert's "*.googleapis.com"
*  issuer: C=US; O=Google Trust Services; CN=Google Internet Authority G3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x379ea8)
> GET / HTTP/1.1
> Host: www.googleapis.com
> User-Agent: curl/7.52.1
> Accept: */*
> 
* Connection state changed (MAX_CONCURRENT_STREAMS updated)!
< HTTP/2 404 
< cache-control: no-cache, no-store, max-age=0, must-revalidate
< pragma: no-cache
< expires: Mon, 01 Jan 1990 00:00:00 GMT
< date: Fri, 14 Jun 2019 12:34:04 GMT
< vary: X-Origin
< vary: Origin,Accept-Encoding
< content-type: text/html; charset=UTF-8
< x-content-type-options: nosniff
< x-frame-options: SAMEORIGIN
< x-xss-protection: 1; mode=block
< server: GSE
< alt-svc: quic=":443"; ma=2592000; v="46,44,43,39"
< accept-ranges: none
< 
* Curl_http_done: called premature == 0
* Connection #0 to host www.googleapis.com left intact
Not Foundpi@raspberrypi:~ $

I also tried with another wifi router and got a different error. I am totally confused now.

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'zero2500-zero250-w48xlq'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 12.1.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:oauth successful
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#configList', 'data': {'entries': [{'value': 'false', 'kind': 'sj#configEntry', 'key': 'canManageFP'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowLocalSuggestions'}, {'value': '3', 'kind': 'sj#configEntry', 'key': 'purchaseAccess'}, {'value': '30000', 'kind': 'sj#configEntry', 'key': 'wsVdtMillis'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'sideloadedPlaybackLogging'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isWarmWelcomeDisabledForWoodstock'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'canCancelSubscription'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'nashvillebg'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isVsAvailable'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'canBrowseSt'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'appstoreRatingPrompt'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellDrawerCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDrawerDownloadMessage'}, {'value': '5000', 'kind': 'sj#configEntry', 'key': 'wsVatMillis'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'nuqRequiredForWsSignup'}, {'value': 'GET MORE', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellNewReleasesCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellListenNowCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'enableWsExtras'}, {'value': 'Subscribe to get millions of songs on-demand, offline, and ad-free', 'kind': 'sj#configEntry', 'key': 'upsellSignedOutMessage'}, {'value': '500', 'kind': 'sj#configEntry', 'key': 'topListMaxSize'}, {'value': '2 skips left', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showLMark'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'conciergeListenNow'}, {'value': '6', 'kind': 'sj#configEntry', 'key': 'minimumAppStartCountForFreeUpsell'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellPreviousCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDow [... cut ...]
DEBUG:gmusicapi.Mobileclient1:GetDeviceManagementInfo(args=[], kwargs={'required_auth': [False, False, True, False]})
DEBUG:gmusicapi.Mobileclient1:{'kind': 'sj#userClientIdList'}
INFO:gmusicapi.Mobileclient1:logged out
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 28, in <module>
    logged_in=api.oauth_login(deviceid)
NameError: name 'deviceid' is not defined

Friday8229, 2019-06-14:
Hi shivasiddharth my assistant started working correctly and even the YouTube is working correctly all I does was deleting the bassist pi folder and going through the installation procedure again and when I commented line 27 and 28 in the media player.py it started working. Thanks for your replies and patience.

Friday8229, 2019-06-14:
Hi shivasiddharth my last question is that can I increase the speed of the assistant like the one In my android phone only in playing songs from YouTube.

shivasiddharth, 2019-06-14:
You cannot control the speed.

commit dbe2ce5f (2019-06-14): Update mediaplayer.py

Fix #909 and other google music issues.
closed 2019-06-14 by shivasiddharth