shivasiddharth/GassistPi #911: Bus error or segmentation error when asking to play spotify playlist

mr2wei, 2019-06-03:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000009c5769c2

 PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Describe the bug:
```
When i ask the assistant to play my playlist in spotify, it plays some random video or gives bus error or segmentation error
[... 290 lines cut ...]
Fatal Python error: Segmentation fault

Thread 0x621ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x649fd470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76f21640 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Segmentation fault
```

### Attach the log file named Gassistpi.log
Log file can be found in /tmp/ directory  
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/3245932/GassistPi.log)

shivasiddharth, 2019-06-04:
Check this custom action section for correct syntax.

mr2wei, 2019-06-04:
When I tried to use the correct syntax, it gives me segmentation error. I have checked previous issues on this and nothing solved my problem so far.

shivasiddharth, 2019-06-05:
Post the terminal output from when you used the correct syntax.

mr2wei, 2019-06-05:
`ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play songs from Spotify"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
Fatal Python error: Segmentation fault

Thread 0x630ff470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x628fe470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76fda640 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Segmentation fault`

this is what happens when i say play songs from spotify

shivasiddharth, 2019-06-05:
Please check the syntax. Correct syntax is using the playlist name, not just "songs".

mr2wei, 2019-06-07:
Oh my playlist name is called songs.

mr2wei, 2019-06-07:
This is what i get when i asked it to play another playlist called 'sleep'

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play sleep from Spotify"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
Fatal Python error: Bus error

Thread 0x641fc470 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 475 in start_detector
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x649fd470 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 248 in pushbutton
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x76fe4640 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 114 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1016 in main
  File "/home/pi/GassistPi/src/main.py", line 1024 in <module>
Bus error
```

mr2wei, 2019-06-07:
i asked it to play my playlist dubstep and it worked the terminal gave me segmentation fault. i cant seem to get it to work again.

aandroide, 2019-06-07:
currently this program works only with its own playlists and not with those shared by other people.
I know that sid is working on this problem and can also reproduce other people's playlists

aandroide, 2019-06-07:
https://github.com/shivasiddharth/GassistPi/issues/469

mr2wei, 2019-06-08:
the playlists i tested are created by me. what do you mean by "own playlists"?

shivasiddharth, 2019-06-11:
@mr2wei Can you post your log file that is generated after the occurrence of the error?

shivasiddharth, 2020-11-02:
Closing this for inactivity.

closed 2020-11-02 by shivasiddharth